autowarefoundation/autoware #1207: how to save map

JSRGFJZ, 2018-04-11:
Hi, I have a new rosbag recorded using velodyne-16, I ran mapping, How can I save the map

YamatoAndo, 2018-04-11:
[image]

1. Click app of ndt_mapping
2. Click "PCD OUTPUT"
If you do not specify a file path, it is saved in < your-home-directory >/.ros/autoware-yymmdd.pcd by default.

JSRGFJZ, 2018-04-11:
well actually I just clone you ndt_localization repo, and run it , is there any way to save the map?

j.z.feng@foxmail.com
 
From: YamatoAndo
Date: 2018-04-11 17:13
To: CPFL/Autoware
CC: Jean; Author
Subject: Re: [CPFL/Autoware] how to save map (#1207)
Click app of ndt_mapping
Click "PCD OUTPUT"
If you do not specify a file path, it is saved in < your-home-directory >/.ros/autoware-yymmdd.pcd by default.

YamatoAndo, 2018-04-11:
Please try the command below:

```
$ source Autoware/ros/devel/setup.bash
$ rostopic pub --once /config/ndt_mapping_output autoware_msgs/ConfigNdtMappingOutput "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
filename: 'test.pcd'
filter_res: 0.2" 

```
saved in < your-home-directory >/.ros/test.pcd

YamatoAndo, 2018-04-11:
It seems you are working in this repository.
https://github.com/JSRGFJZ/SLAM_algorithm

Then, please try the command below:
```
$ source < your-workspace-path >/devel/setup.bash
$ rostopic pub --once /config/ndt_mapping_output runtime_manager/ConfigNdtMappingOutput "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
filename: 'test.pcd'
filter_res: 0.2" 
```

hunkyu, 2019-04-01:
@YamatoAndo How can I save the map in sections?