autowarefoundation/autoware #1221: How to use autoware_driveworks_interface

snuffysasa, 2018-04-19:
Hello,
I am a drive PX developer and I’d like to try using Autoware on the drivePX and interface it with Driveworks libraries. 
I got Autoware installed properly, but I am confused how do I connect with the Driveworks API. 

I see there is an outline here https://github.com/CPFL/Autoware/wiki/DRIVE-PX2-Docker

But it doesn't really explain how to call functions from the driveworks api.

Are there any code samples that show connecting Autoware Driveworks interface package to Driveworks API?

Thanks

amc-nu, 2018-04-25:
@snuffysasa We can't release for the moment Driveworks related API functions publicly. If you're a Driveworks licensee, you can check the source code examples with CMake, those can give you an idea on how to develop your own packages using other Autoware messages and Nvidia proprietary API.

closed 2018-04-25 by amc-nu

snuffysasa, 2018-05-08:
@amc-nu 

Thank you.  I am a driveworks licensee, but I am still quite lost... don't know where to begin.

so I am looking at:
https://github.com/CPFL/Autoware/blob/master/ros/src/driveworks/packages/autoware_driveworks_interface/CMakeLists.txt

In this file would I add something like : 

set(LIBRARIES
    ${Driveworks_LIBRARIES}
    dw_samples_common
    dw_samples_drivenet_common
)

but I also see this:
-lautoware_driveworks

what does this include? all of the driveworks libraries ?  driveworks 0.3 or driveworks 0.6?

Before using Autoware, I tried to figure out how to cross compile the driveworks applications with ROS together, but never got that to work.

swrao, 2018-06-12:
@snuffysasa  

Were you able to use Autoware with driveworks.  I am a driveworks licensee too, please let me know if you were successful.