autowarefoundation/autoware #1228: Hello Everyone !

ptiwari0664, 2018-04-24:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

amc-nu, 2018-04-25:
Gazebo is not currently supported in 16.04

closed 2018-04-25 by amc-nu

ptiwari0664, 2018-04-30:
Thanks for the response. Will it work with Wily (Ubuntu 15.10) ???

Thanks in advance !!!

rahulbhadani, 2019-04-03:
It will be great, if you post the issue to the original catvehicle repository at https://github.com/sprinkjm/catvehicle. Developers of catvehicle will be able to address it better.