autowarefoundation/autoware #1232: "rosrun runtime_manager runtime_manager_dialog.py "   error

vehicularkech, 2018-04-25:
When i run "rosrun runtime_manager runtime_manager_dialog.py ",error happends and no GUI Woindow show:

/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2862: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()
loading param.yaml
loading qs.yaml
Traceback (most recent call last):
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 3416, in <module>
    app = MyApp(0)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2863, in OnInit
    frame_1 = MyFrame(None, wx.ID_ANY, "")
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 147, in __init__
    self.setup_buttons(self.qs_dic.get('buttons', {}), self.qs_cmd)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 725, in setup_buttons
    pdic = self.load_dic_pdic_setup(k, d2)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 1978, in load_dic_pdic_setup
    pdic = self.load_dic.get(name, {})
AttributeError: 'NoneType' object has no attribute 'get'
Unhandled exception in thread started by <bound method Thread.__bootstrap of <Thread(Thread-3, stopped daemon 547784487424)>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 774, in __bootstrap
    self.__bootstrap_inner()
  File "/usr/lib/python2.7/threading.py", line 814, in __bootstrap_inner
    (self.name, _format_exc()))
  File "/usr/lib/python2.7/traceback.py", line 241, in format_exc
    etype, value, tb = sys.exc_info()
AttributeError: 'NoneType' object has no attribute 'exc_info'

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: master

ncnynl, 2018-07-29:
May be you don't install  wx for windows .   try   sudo apt-get install python-wxtools

hamkuchan, 2018-09-13:
@vehicularkech I met the same issue, have you fixed it?

vehicularkech, 2018-09-18:
By reinstalling Autoware, i sloved this problem

gnykumar, 2019-05-10:
Everything was working well.
Suddenly I am also facing same issue, without uninstalling or installing anything.

`/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2889: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()
loading param.yaml
loading qs.yaml
Traceback (most recent call last):
  File "/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 3443, in <module>
    app = MyApp(0)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2890, in OnInit
    frame_1 = MyFrame(None, wx.ID_ANY, "")
  File "/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 131, in __init__
    self.setup_buttons(self.qs_dic.get('buttons', {}), self.qs_cmd)
  File "/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 709, in setup_buttons
    pdic = self.load_dic_pdic_setup(k, d2)
  File "/home/User/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 1963, in load_dic_pdic_setup
    pdic = self.load_dic.get(name, {})
AttributeError: 'NoneType' object has no attribute 'get'`

gbiggs, 2019-05-12:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-05-12 by gbiggs

sumenpuyuan, 2019-12-18:
> By reinstalling Autoware, i sloved this problem

Hello! Is reinstallation a recompilation?
just ：   ./catkin_make_release  ？

sumenpuyuan, 2019-12-18:
> @vehicularkech I met the same issue, have you fixed it?

Hello! How did you solve your last problem?

CCTJB2, 2020-09-15:
I found out how to solve it in my case, linked to another thread which describes the same problem: 
https://github.com/Autoware-AI/autoware.ai/issues/1508#issuecomment-692506922