autowarefoundation/autoware #1244: Gazebo CMD_VEL (Slipping on Terrain)

ptiwari0664, 2018-05-07:
Hello Everyone!

I am trying to simulate catvehicle in gazebo. After a successful installation and with launching:
roslaunch catvehicle catvehicle_skidpan.launch (urban.world) and gzclient, I am clearly able to see the vehicle in the world.
After this I published a command velocity of linear.x = 3.0 to the topic /catvehicle/cmd_vel or /catvehicle/cmd_vel_safe 
**rostopic pub  /catvehicle/cmd_vel geometry_msgs/Twist "linear:   
  x: 3.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" -r 10**

# Expected Behavior
The vehicle is able to move but the movement should be fast as, its around 10KMPH.

# Actual Behavior
The vehicle movement is really slow around 0.6 m/s. 

#Specification 
Ubuntu version: 14.04 LTS
ROS Version: Indigo
Gazebo version: 7.0

Thanks in Advance !!!!!

MrDadaGuy, 2018-07-27:
I have this same issue, although I am on 16.04 and kinetic.  Same symptoms.  @ptiwari0664 , have you had any breakthrough?  Thanks

carolina68, 2018-10-15:
I have also got the same problem. Is it related to simulation physics or something else? Thanks in advance if anyone has got the answer.

yukkysaito, 2019-02-16:
Old gazebo will be replace the new Gazebo.
This issue will be closed.
#1930

closed 2019-02-22 by yukkysaito