autowarefoundation/autoware #1246: Issue while building Autoware on Aarch64 host

Mani-Sadhasivam, 2018-05-07:
# Bug

Building Autoware on Aarch64 hardware throws below error:

```shell
/usr/bin/ld: /home/linaro/Autoware/ros/src/actuation/vehicles/packages/ymc/lib/libymc_can_2.0_aarch64.a(cansend.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against external symbol `__stack_chk_guard@@GLIBC_2.17' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /home/linaro/Autoware/ros/src/actuation/vehicles/packages/ymc/lib/libymc_can_2.0_aarch64.a(cansend.cpp.o)(.text+0x30): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `__stack_chk_guard@@GLIBC_2.17'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
actuation/vehicles/packages/ymc/CMakeFiles/g30esli_interface.dir/build.make:114: recipe for target '/home/linaro/Autoware/ros/devel/lib/ymc/g30esli_interface' failed
make[2]: *** [/home/linaro/Autoware/ros/devel/lib/ymc/g30esli_interface] Error 1
CMakeFiles/Makefile2:35251: recipe for target 'actuation/vehicles/packages/ymc/CMakeFiles/g30esli_interface.dir/all' failed
```
Should the library `libymc_can_2.0_aarch64.a` compiled as a shared object?

## Specifications

  - Debian version: Stretch
  - ROS Version: Kinetic
  - Autoware branch: Master

yk-fujii, 2018-05-08:
Thanks for report!

Yes, `libymc_can_2.0_aarch64.a` has compiled as a shared object.
Unfortunately, this library has a rights problem, we can not publish the source code. 

If you don't use yamaha's golfcart controller, please remove `ros/src/actuation/vehicles/packages/ymc`.

amc-nu, 2018-05-08:
or add a CATKIN_IGNORE file inside `ros/src/actuation/vehicles/packages/ymc`

Mani-Sadhasivam, 2018-05-08:
Hi @yk-fujii @amc-nu,

Thanks for your inputs. I'm able to build Autoware on ARMv8 host successfully after excluding several packages!

Also I have faced several compilation errors during building for which I have submitted fix #1249.

closed 2018-05-08 by Mani-Sadhasivam