autowarefoundation/autoware #1261: Autoware /Docker   installation  to DrivePX2

Todspaniard, 2018-05-10:
We are DrivePX2  algorithm developper  and trying  to  install  Autoware (Docker)  .
After installation finished with sucsessfully ,  we got  a error masage and
drive works related function (GMSL camera input  or  Lidar point cloud data )
can not work.
It's  seems that  Autoware can not find out driveworks APi at all.

Could you suggest us how to resolve the issue .

yk-fujii, 2018-05-11:
Could you tell me the execution steps and full error messages?

yk-fujii, 2018-05-11:
Please also tell me the version of Driveworks.

Todspaniard, 2018-05-11:
Thanks a lot for your quick response.

1.Driveworks version  is  driveworks-0.6.(SDK 5.0.5.0)
2.Excution steps are following.
  /install step is  as   "https://github.com/CPFL/Autoware/wiki/DRIVE-PX2-Docker"   .
 /excution  step  is
      $ autoware-docker.
      $ ~/start_autoware.sh

yk-fujii, 2018-05-11:
Okey, in that state only RuntimeManager is up.
Did you run `roslaunch autoware_driveworks_interface gmsl_interface.launch` ?
```
cd autoware/ros
source devel/setup.bash
roslaunch autoware_driveworks_interface gmsl_interface.launch
```
Also, you needs to operate the RuntimeManager for starting the driver.

Todspaniard, 2018-05-12:
Deer Sir,
We tried flowing  excution process  but reslove it.

$ autoware-docker
[INFO]:Autoware-Driveworks module is found. Autoware can use driveworks.
[WARN]:Execute script with sudoers.
[sudo] password for nvidia: 
[INFO]:The autoware-docker image [tier4cr.io/dpx2/autoware:1.6.2] was ready.
[INFO]:GMSL Camera initialization sequence is start.
.....autoware-docker: line 138: kill: (6604) - No such process

[INFO]:GMSL Camera initialization sequence was end.
ls: cannot access '/media/nvidia': No such file or directory
non-network local connections being added to access control list
access control disabled, clients can connect from any host
nvidia@nvidia:~$ cd autoware/ros
nvidia@nvidia:~/autoware/ros$ source devel/setup.bash
nvidia@nvidia:~/autoware/ros$ roslaunch autoware_driveworks_interface gmsl_interface.launch
[gmsl_interface.launch] is neither a launch file in package [autoware_driveworks_interface] nor is [autoware_driveworks_interface] a launch file name
The traceback for the exception was written to the log file

yk-fujii, 2018-05-14:
Does following directory exist in your image?
`~/autoware/ros/src/driveworks/packages/autoware_driveworks_interface/launch`

Please also refer to the videos below.
### how to autoware with docker+px2
https://www.youtube.com/watch?v=3UO1K0lD2Qk
### how to use gmsl with docker+px2
https://www.youtube.com/watch?v=-swTFYO3Fuc

Todspaniard, 2018-05-14:
detail error message was bellow.
nvidia@nvidia:~/autoware/ros$ rosrun autoware_driveworks_interface gmsl_interface 
Initialize DriveWorks SDK v0.6.67
Release build with GNU 4.9.2 from v0.6.0-rc7-0-g51bd3aa against Vibrante PDK v5.0.5.0
TimeSource: monotonic epoch time offset is 1526266051951033
TimeSource: PTP ioctl returned error. Synchronized time will not be available.
TimeSource: Could not detect valid PTP time source at 'eth0'. Fallback to CLOCK_MONOTONIC.
SDK: Resources mounted from /usr/local/driveworks-0.6/data/resources
nvrm_gpu: Bug 200215060 workaround enabled.
SDK: Create NvMediaDevice
egl::Display: found 2 EGL devices
egl::Display: use drm device: drm-nvdc
SDK: number of GPU devices detected 2
SDK: currently selected GPU device discrete ID 0
SAL: identified board as DrivePX2-TegraA
SensorFactory::createSensor() -> camera.gmsl, camera-type=ar0231-rccb,csi-port=ab,fifo-size=3,required-framerate=20,slave=0,output-format=yuv
CameraGMSL: required FPS = 20, resolution = 1920x1208
iscRootDevOpen: can't open /dev/isc-mgr.7.a - Device or resource busy
NvMediaISCRootDeviceCreate: Unable to open root device
nvmedia: ERROR: Init: Failed to create NvMedia ISC root device
Driveworks exception thrown: DW_SAL_CANNOT_CREATE_SENSOR: CameraGMSL: cannot connect to camera

Cannot create driver: camera.gmsl with params: camera-type=ar0231-rccb,csi-port=ab,fifo-size=3,required-framerate=20,slave=0,output-format=yuv
Error: DW_SAL_CANNOT_CREATE_SENSOR

yk-fujii, 2018-05-14:
ah, current driver setting fixed parameter is `ar0231-rccb,csi-port=ab,fifo-size=3,required-framerate=20,slave=0,output-format=yuv`.

It's a very bad implementation...;(
Also, I am planning to fix some bugs and implementation problems, 
but I think that we can merge immediately.

Could you tell me the parameter for your camera(ov10640)?
I'll send you a libautoware_driveworks.so to set your parameter.

Todspaniard, 2018-05-14:
for  driveworks  API, we are using following comand and parameter ,that works well.
./sample_camera_gmsl --csi-port=ab --camera-type=ov10635

and   lidar ,
./sample_lidar_replay --protocol=lidar.socket --params=device=VELO_VLP16,ip=192.168.1.201,port=2368,scan-frequency=12.5 --show-intensity=true

that's is also works well .

those are our environment.

Todspaniard, 2018-05-14:
also   for recording of  driveworks  API  , we use 
"params": "camera-type=ov10635,csi-port=ab,camera-count=1,output-format=yuv,async-record=1,fifo-size=6"

closed 2020-05-29 by gbiggs