autowarefoundation/autoware #1267: ndt_matching doesn't work properly on GPU on Jetson TX2

TanatosSpirit, 2018-05-14:
# Bug
"ndt_mapping" and "ndt_matching" doesn't work with flag "Use_gpu" on arm64 systems.

## Actual Behavior
"ndt_mapping" and "ndt_matching" work properly on laptop using CUDA(I've tested it on intel i7 + GTX 1050). Unfortunately, when I tried to launch it on Jetson TX2 using CUDA, the results weren't OK. 
I've applied the fix described in issue #1139. Ndt-matching starts, but results of calculation aren't correct.

## Steps to Reproduce the Problem

  1. Use ndt_maching with flag "Use_gpu" on Jetson TX2

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: master
  - Machine: Nvidia Jetson TX2
  - CPU: aarch64
  - Host and GPU mem (shared): 8 Gb

yk-fujii, 2018-05-15:
This problem seems to be a problem commonly occurring in Tegra's CUDA.
#1268 is the same topic, so I would like to talk at #1268

anhnv3991, 2018-05-15:
@TanatosSpirit can you show me the result on Tegra?

TanatosSpirit, 2018-05-15:
@anhnv3991 of course. What do you exactly want to see: log of ndt_matching, screenshot of result in rviz or all in one?

anhnv3991, 2018-05-15:
@TanatosSpirit please send me all: screenshot of the matching process on rviz and the log file.

snuffysasa, 2018-05-15:
I am having this same issue on Drive PX 2

TanatosSpirit, 2018-05-16:
@anhnv3991 no, it didn't work properly. I meant ndt_matching didn't start with block_size_x 1024 and returned an error. block_size_x had been changed to 512 before recording videos. This bug is described in #1274.

anhnv3991, 2018-05-16:
@snuffysasa What is your problem on Drive PX2? Your ndt_matching did not launch? Or it launched but did not run properly?

anhnv3991, 2018-05-16:
@TanatosSpirit Well, then it is more complicated than I thought. Have you tried to reduce BLOCK_SIZE_X more (to 256 or 128)?
Btw, have you tried ndt_mapping? Did it work properly on Tegra X2?

snuffysasa, 2018-05-16:
@anhnv3991 

It launches fine.

It works... however, something seems wrong with the NDT matching now.. the ndt matching is unstable and not staying aligned properly... and this does not happen when using CPU ndt matching on drive px 2 with the same data.

And this also does not happen, when using the same data on Desktop PC CPU and GPU ndt matching.

TanatosSpirit, 2018-05-16:
@snuffysasa so, do you use BLOCK_SIZE_X = 1024 to launch ntd_matching?

TanatosSpirit, 2018-05-16:
@anhnv3991 yes, I've changed BLOCK_SIZE_X (512,256, 128) and ndt_matching still doesn't work properly (like on vidoes). Ndt_mapping works just like ndt_matching.
I'll try to switch to develop branch.

anhnv3991, 2018-05-16:
@TanatosSpirit Please tell me wheather the develop branch works on your system or not. I think there is a problem in the NormalDistributionTransform.cu file, which is already fixed in the develop branch. But I am not sure. I need to know how the develop branch work on Tegra X2.

anhnv3991, 2018-05-16:
@TanatosSpirit Ok, thank you for your help. The BLOCK_SIZE_X problem is clear, I will fix it soon (maybe by adding a ifdef to define BLOCK_SIZE_X on different arch).
However, it is just a minor problem. The bigger problem is develop branch does not work properly on your system. I worked on my GTX Titan and we also tested it on Drive PX2 as well.
Let's wait for @snuffysasa and see if develop branch works on his DrivePX2 or not.
Anyway, I will talk to my team to see if they have any TegraX2 so we can figure out what actually happen. 

@snuffysasa When you try develop branch on DrivePX2, could you please keep BLOCK_SIZE_X at 1024?

snuffysasa, 2018-05-16:
@TanatosSpirit @anhnv3991 

Okay i will try develop branch now. Can you be clear what branch exactly? link?

Also, I will keep BLOCK_SIZE_X at 1024

snuffysasa, 2018-05-16:
@TanatosSpirit  @anhnv3991 

Sorry for delay. Had some troubles getting develop branch to work.

But now, I do have develop branch working. 

The issues with GPU not_matching is still there!

On master branch and develop branch on Desktop PC
Every ndt matching mode works

On develop branch on the Drive Px2

- pcl generic works
- pcl anh works
- pcl gpu has the issue
- pcl openmp works

on master branch on drive PX2
- default works
- with openMP  works
- with fast_pcl works
- with gpu has the issue.

And the issue is, it is not crashing or anything.. and it does seem to align correctly, but then it makes strange jumps off of the alignment after it has been aligned.

anhnv3991, 2018-05-17:
@snuffysasa Thank you. I think I know what the problem is. Maybe it comes from the file NormalDistributionTransform.cu. 
Fortunately, I have a DrivePx2 at my Lab now. After I fix the bug, I will let you and @TanatosSpirit know. Please wait for a while.

anhnv3991, 2018-05-17:
@snuffysasa @TanatosSpirit  try this:
https://github.com/CPFL/Autoware/tree/fix/gpu_ndt_drivePX2

TanatosSpirit, 2018-05-17:
@anhnv3991 @snuffysasa  Ndt_matching works properly! @anhnv3991 Thank you very much! I tested with BLOCK_SIZE_X at 512 and 256 and i didn't see the difference. 
But logs is a bit strange. Log file has not correct flags such as gnss_use and openMP_use. I did't use the gnss.
[ndt_matching-1-stdout.log](https://github.com/CPFL/Autoware/files/2013144/ndt_matching-1-stdout.log)
[ndt_matching_omp-1-stdout.log](https://github.com/CPFL/Autoware/files/2013145/ndt_matching_omp-1-stdout.log)
And the GPU's "Fitness Score" is two orders of magnitude more than the OpenMP's "Fitness Score" while view in rviz is the same.
Of course, Tegra X2 has less perfomance than the GP106. Maybe you know some way to reduce GPU load?

anhnv3991, 2018-05-17:
@TanatosSpirit glad to hear it.
Fitness Score is the average distance (in meters) between each scanned point (points produced by the car's scanner) and its nearest point on the map. It is used to avoid the cases of miss-align (such as the location of the car determined by matching process is 100 meters far from its actual location).
Fitness score depends on the searching method, which is used to search for the nearest neighbor of each scanned point (such as kd-tree, octree, etc). I used octree for my implementation, but initially it is very slow, so I had to adjust it a bit. As a result, it becomes faster, but less accurate. That's why you see the fitness score of gpu is about 5 meters, while on openMP it is just 0.05 meters.
I am working on a more accurate searching method, but it has not completed yet.

As for reducing GPU Load, I'm sorry but I don't know any way to do it.

TanatosSpirit, 2018-05-17:
@anhnv3991 thank you for explanation.

snuffysasa, 2018-05-17:
@anhnv-3991 
@TanatosSpirit 

This fix does not work on my drive px 2

(runtime_manager_dialog.py:27738): Gtk-WARNING **: Unable to show 'app': Operation not supported
['roslaunch', 'ndt_localizer', 'ndt_matching.launch', 'use_openmp:=False', 'use_fast_pcl:=False', 'use_gpu:=True', 'get_height:=False']
pid=8055
sched policy=OTHER prio=0
ndt_matching: /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:838: Eigen::internal::block_evaluator<ArgType, BlockRows, BlockCols, InnerPanel, true>::block_evaluator(const XprType&) [with ArgType = const Eigen::Matrix<double, 6, 6>; int BlockRows = 6; int BlockCols = -1; bool InnerPanel = true; Eigen::internal::block_evaluator<ArgType, BlockRows, BlockCols, InnerPanel, true>::XprType = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>]: Assertion `((size_t(block.data()) % (((int)1 >= (int)evaluator<XprType>::Alignment) ? (int)1 : (int)evaluator<XprType>::Alignment)) == 0) && "data is not aligned"' failed.
[ndt_matching-1] process has died [pid 8074, exit code -6, cmd /home/nvidia/Autoware/ros/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/nvidia/.ros/log/87e10938-59e4-11e8-ae1b-00044b707976/ndt_matching-1.log].
log file: /home/nvidia/.ros/log/87e10938-59e4-11e8-ae1b-00044b707976/ndt_matching-1*.log
Unhandled exception in thread started by
... logging to /home/nvidia/.ros/log/87e10938-59e4-11e8-ae1b-00044b707976/roslaunch-nvidia-8055.log

anhnv3991, 2018-05-17:
@snuffysasa The link I sent you is for master branch. Where did you patch it to?

snuffysasa, 2018-05-17:
@anhnv3991 

I just checked out to that branch,  and looks like its a fork from master branch

TanatosSpirit, 2018-05-17:
@snuffysasa   It seems I had a similar error some months ago. I changed a code or a file and built only ndt_localizer package using 
`$ catkin_make --pkg ndt_localizer`
Then I built all autoware using
`./catkin_make_release`
and error didn't appear.

snuffysasa, 2018-05-17:
@TanatosSpirit 
I'll try this, but I am confused... why compile the single package.. if you are going to recompile all of it?

TanatosSpirit, 2018-05-17:
@snuffysasa sorry, I meant that compiling the single package didn't work. Need to recompile all.

snuffysasa, 2018-05-17:
No luck.  Recompiled all of autoware several times, trying various things. 

No matter what now, I can't get ndt_matching to work on drive Px with fast_pcl, with or without gpu.  Only can get the regular ndt_matching to work.

But earlier this week I could get cpu fast_pcl to work.  

and none of the ndt_mathcing works in develop branch

snuffysasa, 2018-05-18:
@anhnv3991  @TanatosSpirit 

I finally got it.  It was a long fight.

I got it to work in docker container, but had to reinstall the image.  It's working now and I'm happy.

Thanks to both of you for the help

anhnv3991, 2018-05-18:
@snuffysasa Basically, in ndt_matching (and also in ndt_mapping), we approximate the map by breaking it into many voxels (put the map into a voxel grid). The smaller the size of each voxel is, the more voxel the map will produce. Hence, if you use large map and small-size voxel, the GPU can easily run out of memory.

So, you can reduce the number of voxels by increasing their size (which is called 'resolution') as follow:
1. From the runtime manager, selecting 'app' button from the ndt_matching.
2. In the app button, increase the resolution (in my GTX Titan, I set resolution by 3).

As for igpu or dgpu, sorry but I don't know. This code is made on my PC, then run on DrivePX2, so the DrivePX2 decides which gpu is better to run the code.

snuffysasa, 2018-05-18:
@anhnv3991 

Okay cool. That's what I figured.  

at some point I will explore what GPU on the drivePX is being used, because that could make a big difference.

closed 2018-06-20 by anhnv3991