autowarefoundation/autoware #1268: Nvidia Drive PX2 NDT matching GPU error

snuffysasa, 2018-05-14:
# Bug
## Expected Behavior
GPU performs NDT matching node

## Actual Behavior
Error: operation not supported /home/nvidia/autoware/catkin_autoware/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/Registration.cu 366

## Steps to Reproduce the Problem

  1. compile autoware master branch on nvidia drive px 2... Cuda 8
  1. load autoware localization demo with ndt... cpu mode it works.
  1. check "use gpu"  then run ndt matching

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch: master

this is line 366 in registration.cu

checkCudaErrors(cudaHostRegister(host_tmp, sizeof(pcl::PointXYZ) * target_points_number_, cudaHostRegisterDefault));

I think it has to do with cudaHostRegister()

snuffysasa, 2018-05-15:
@anhnv3991 

Thank you for the help!

I tried the fix in #1139 ... and it worked! on my drivePx2 ! thanks. and great news!

However,  it works... however, something seems wrong with the NDT matching now..  the ndt matching is unstable and not staying aligned properly... and this does not happen when using CPU ndt matching on drive px 2.

And this also does not happen, when using the same data. on Desktop PC CPU and GPU ndt matching.

Any ideas on this?

thanks

snuffysasa, 2018-05-15:
Update:

I think my issue must be the same as issue #1267

anhnv3991, 2018-05-16:
@snuffysasa yes I did, so I appreciate if we could discuss more about this problem in #1267.

closed 2018-05-21 by kitsukawa