autowarefoundation/autoware #1276: Runtime Manager does not display (no module autoware_msgs.msg)

milanow, 2018-05-16:
Hi I am kinda new to python and linux. My system is Ubuntu 16.04

Here is the problem, when I try ./run to test out Autoware, the process manager doesn't display. And I found the issue #919 , which is similar with my problem. However, when I run command "source devel/setup.bash; roscore & rosrun runtime_manager runtime_manager_dialog.py", iit reports "ImportError: No module named autoware_msgs.msg". I think my question is clear, where should I install/find out the module?

Thanks

milanow, 2018-05-16:
Actually, when building Autoware, this issue has happend:

CMake Error at /opt/ros/kinetic/share/catkin/cmake/assert.cmake:20 (message):
Assertion failed: file
  '~/Autoware/ros/src/msgs/autoware_msgs/msg/autoware_msgs.msg'
  does not exist.  Message: message file not found

kitsukawa, 2018-05-21:
@milanow Did you compile Autoware using `catkin_make_release` before running `run`?

lijecaru, 2018-06-11:
@kitsukawa I have the same issue. When I try to run the runtime_manager, I get the error: ImportError: No module named autoware_msgs.msg. When I run catkin_make I get a CMake Error, it says: Could not find Glew (missing: GLEW_INCLUDE_DIR GLEW_LIBRARY). I checked #682, but I have python installed (it doesn't seem to be the problem)

jraclarke, 2018-06-12:
@lijecaru would running the following and then recompiling work?

`sudo apt-get install libglew-dev`

lijecaru, 2018-06-12:
Thank you @jraclarke. I ran the command and then use `catkin_make` in the directory ~/Autoware/ros, but I get a Cmake errror (couldn't find a package configuration file provided by "module_comm_msgs"). If I try to run the Runtime manager again, I get the same error as before: No module named autoware_msgs.msg.

Should I install Autoware again somehow? What else could I have missed?

yk-fujii, 2018-06-13:
@lijecaru 
I guess you did not clone submodule .
please try `git submodule update --init` and retry `catkin_make_release`

AthpapAutowareUser, 2018-06-23:
Hi. I am actually facing the exact same issue.
I am not able to see the Runtime Manager and I get the same error message: 
Command:    cd $HOME/Autoware/ros/; source devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py
Message: Traceback (most recent call last):
  File "/home/autoware/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 59, in <module>
    from autoware_msgs.msg import ConfigSsd
ImportError: No module named autoware_msgs.msg
The catkin_make_release was not run without any problems. I still get : 
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "jsk_recognition_msgs" with any of the following names:

    jsk_recognition_msgsConfig.cmake
    jsk_recognition_msgs-config.cmake

  Add the installation prefix of "jsk_recognition_msgs" to CMAKE_PREFIX_PATH
  or set "jsk_recognition_msgs_DIR" to a directory containing one of the
  above files.  If "jsk_recognition_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  msgs/autoware_msgs/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'jsk_recognition_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "jsk_recognition_msgs" with any of the following names:

    jsk_recognition_msgsConfig.cmake
    jsk_recognition_msgs-config.cmake

  Add the installation prefix of "jsk_recognition_msgs" to CMAKE_PREFIX_PATH
  or set "jsk_recognition_msgs_DIR" to a directory containing one of the
  above files.  If "jsk_recognition_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  msgs/autoware_msgs/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

I have tried all the above suggestion any nothing changed.
Can anyone help my here?

Tikitikitiki, 2018-07-06:
try sudo apt-get install ros-kinetic-jsk-recognition-msg

yuqli, 2018-11-06:
> try sudo apt-get install ros-kinetic-jsk-recognition-msg

This results in `E: Unable to locate package ros-kinetic-jsk-recognition-msg` on my system.

AlexeiFedorov, 2018-11-22:
> try sudo apt-get install ros-kinetic-jsk-recognition-msg
This results in E: Unable to locate package ros-kinetic-jsk-recognition-msg on my system.

This should be actually
sudo apt-get install ros-kinetic-jsk-recognition-msgs
(msgs instead of msg)

pedrobagatin, 2019-02-07:
> > try sudo apt-get install ros-kinetic-jsk-recognition-msg
> 
> This results in `E: Unable to locate package ros-kinetic-jsk-recognition-msg` on my system.

ros-kinetic-jsk-recognition-msgs

closed 2020-05-29 by gbiggs