autowarefoundation/autoware #1277: [Fix] waypoint_loader doesn't work from QuickStart

kfunaoka, 2018-05-17:
# Bug
waypoint_loader doesn't work from Quick Start even if my_launch for the moriyama sample is fixed.

## Expected Behavior
waypoint is displayed.

## Actual Behavior
waypoint is not displayed.

## Steps to Reproduce the Problem

  1. Apply patch to the files generated by my_launch.sh
```
diff --git a/my_detection.launch b/my_detection.launch
index cf58222..e8fd72d 100644
--- a/my_detection.launch
+++ b/my_detection.launch
@@ -21,37 +21,37 @@
   <node pkg="points2image" type="points2image" name="points2image" />
 
   <!-- car and pedestrian detection -->
-  <!-- dpm_XXX -->
-  <include file="$(find cv_tracker)/launch/dpm_ttic.launch">
+  <!-- dpm_ttic -->
+  <include file="$(find vision_dpm_ttic_detect)/launch/vision_dpm_ttic_detect.launch">
     <arg name="car" value="$(arg car_detection)" />
     <arg name="pedestrian" value="$(arg pedestrian_detection)" />
     <arg name="use_gpu" value="$(arg is_use_gpu)" />
   </include>
 
   <!-- range_fusion -->
-  <include file="$(find cv_tracker)/launch/ranging.launch">
+  <include file="$(find range_fusion)/launch/range_fusion.launch">
     <arg name="car" value="$(arg car_detection)" />
     <arg name="pedestrian" value="$(arg pedestrian_detection)" />
   </include>
 
-  <!-- XXX_track -->
-  <include file="$(find cv_tracker)/launch/klt_tracking.launch">
+  <!-- vision_klt_track_track -->
+  <include file="$(find vision_klt_track)/launch/vision_klt_track.launch">
     <arg name="car" value="$(arg car_detection)" />
     <arg name="pedestrian" value="$(arg pedestrian_detection)" />
   </include>
 
   <!-- obj_reproj -->
-  <include file="$(find cv_tracker)/launch/reprojection.launch">
+  <include file="$(find obj_reproj)/launch/obj_reproj.launch">
     <arg name="car" value="$(arg car_detection)" />
     <arg name="pedestrian" value="$(arg pedestrian_detection)" />
   </include>
 
-  <!-- euclidean_cluster -->
-  <include file="$(find lidar_tracker)/launch/euclidean_clustering.launch">
+  <!-- lidar_euclidean_cluster_detect -->
+  <include file="$(find lidar_euclidean_cluster_detect)/launch/lidar_euclidean_cluster_detect.launch">
   </include>
 
   <!-- obj_fusion -->
-  <include file="$(find lidar_tracker)/launch/obj_fusion.launch">
+  <include file="$(find obj_fusion)/launch/obj_fusion.launch">
     <arg name="car" value="$(arg car_detection)" />
     <arg name="pedestrian" value="$(arg pedestrian_detection)" />
   </include>
@@ -60,12 +60,12 @@
   <!-- traffic light recognition -->
   <!-- feat_proj -->
   <!--
-  <node pkg="road_wizard" type="feat_proj" name="feat_proj" />
+  <node pkg="trafficlight_recognizer" type="feat_proj" name="feat_proj" />
   -->
 
   <!-- region_tlr -->
   <!--
-  <include file="$(find road_wizard)/launch/traffic_light_recognition.launch" />
+  <include file="$(find trafficlight_recognizer)/launch/traffic_light_recognition.launch" />
   -->
 
 </launch>
diff --git a/my_localization.launch b/my_localization.launch
index dc874af..dba0166 100644
--- a/my_localization.launch
+++ b/my_localization.launch
@@ -25,8 +25,7 @@
   <include file="$(find gnss_localizer)/launch/nmea2tfpose.launch"/>
 
   <!-- ndt_matching -->
-  <include file="$(find ndt_localizer)/launch/ndt_matching.launch">
-    <arg name="use_openmp" value="$(arg use_openmp)" />
+  <include file="$(find lidar_localizer)/launch/ndt_matching.launch">
     <arg name="get_height" value="$(arg get_height)" />
   </include>
 
diff --git a/my_sensing.launch b/my_sensing.launch
index 9cb948f..205b847 100644
--- a/my_sensing.launch
+++ b/my_sensing.launch
@@ -13,7 +13,7 @@
   <!-- <node pkg="javad_navsat_driver" type="gnss.sh" name="javad_driver"/> -->
 
   <!-- PointGrey Grasshopper3 -->
-  <include file="$(find pointgrey)/scripts/grasshopper3.launch">
+  <include file="$(find autoware_pointgrey_drivers)/scripts/grasshopper3.launch">
     <arg name="CalibrationFile" value="$(arg camera_calib)"/>
   </include>
```

  1. Do just like Quick Start video
      https://www.youtube.com/watch?v=NDNcy0C-Has

## Specifications

  - Ubuntu version: 16.04.4 LTS (Xenial Xerus)
  - ROS Version: Kinetic
  - Autoware branch: release/1.7.0

kfunaoka, 2018-05-18:
Fixed at Commit 0688f39

closed 2018-05-18 by kfunaoka

7chenko, 2019-02-27:
The waypoint loader again doesn't work in QuickStart. I'm following the video here https://www.youtube.com/watch?v=OWwtr_71cqI

All prior processes work, but running the my_mission_planning.launch script, it errors with:

`[config_waypoint_loader_rostopic-4] process has died [pid 14291, exit code 1, cmd /opt/ros/kinetic/lib/rostopic/rostopic pub -l /config/waypoint_loader autoware_msgs/ConfigWaypointLoader {multi_lane_csv: /home/autoware/.autoware/data/path/moriyama_path.txt, replanning_mode: false, velocity_max: 20.0, velocity_min: 4.0, accel_limit: 0.98, decel_limit: 0.98, radius_thresh: 20.0, radius_min: 6.0, resample_mode: true, resample_interval: 1.0, velocity_offset: 4, end_point_offset: 5} __name:=config_waypoint_loader_rostopic __log:=/home/autoware/.ros/log/9f0af6d2-358f-11e9-ad77-9cb6d0fc87d3/config_waypoint_loader_rostopic-4.log].
log file: /home/autoware/.ros/log/9f0af6d2-358f-11e9-ad77-9cb6d0fc87d3/config_waypoint_loader_rostopic-4*.log`

And the log is empty. The referenced file moriyama_path.txt exists.

Thanks for your help!