autowarefoundation/autoware #1298: E: Unable to locate package ros--catkin

NEUdeep, 2018-05-27:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

sudo apt-get install -y ros-$ROS_DISTRO-catkin
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros--catkin

## Specifications

  - Ubuntu version:16.04
  - ROS version:
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

NEUdeep, 2018-05-27:
i know,it is about ros that not install in my host.

amc-nu, 2018-05-31:
glad you found the solution and shared. thanks

closed 2018-05-31 by amc-nu

AthpapAutowareUser, 2018-06-23:
Sorry, but how did you fixed the problem? I am still struggling with that

Srivatsa-M-Hegde, 2018-10-17:
@NEUdeep can you please elaborate on how you fixed this issue ?

amc-nu, 2018-10-17:
As the error indicates the $ROS_DISTRO environment var is not set.

Check environment setup  in the ROS installation

I.e. kinetic

echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
source ~/.bashrc

amc-nu, 2018-10-17:
run:
$ sudo apt-key update
$ sudo apt-get update
then 
$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO

rkoyama1623, 2019-10-13:
I had the same problem.
```bash
$ sudo -H apt-get install -y ros-kinetic-nlopt
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  ros-kinetic-nlopt
0 upgraded, 1 newly installed, 0 to remove and 174 not upgraded.
Need to get 0 B/178 kB of archives.
After this operation, 523 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  ros-kinetic-nlopt
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```
I do **not** think the following solution is good, but I was able to avoid the error.
### Temporal Solution:
Open the `/etc/apt/apt.conf`, and add following commands.
```
APT { Get { AllowUnauthenticated "1"; }; }; 
```

### My Environment
```bash
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.6 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.6 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

JeremyCKT, 2022-03-27:
> As the error indicates the $ROS_DISTRO environment var is not set.
> 
> Check environment setup in the ROS installation
> 
> I.e. kinetic
> 
> echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc source ~/.bashrc

But when I do this, there's another error occured.
bash: /opt/ros/kinetic/setup.bash: No such file or directory