autowarefoundation/autoware #1299: Ray ground filter logic

Tamme, 2018-05-28:
# Bug
I'm analysing the code of ray ground filter on master branch and one of the checks doesnt make sense.

Lines 174-176
`          if(current_height <= (-sensor_height_ + general_height_threshold)
              && current_height >= (-sensor_height_ - general_height_threshold))
          {..}
`
And in lines 190-194
` 
        if (points_distance > reclass_distance_threshold_ &&
            (current_height <= (-sensor_height_ + height_threshold)
             && current_height >= (-sensor_height_ - height_threshold))
           )
        {...}
` 

I would think that in both places height should be compared with the global threshold.

@amc-nu, what do you think?

amc-nu, 2018-05-28:
general threshold is a fallback option (given by reclass_distance_threshold).
checking always against the global threshold would make checking consecutives points in the ray pointless.

Tamme, 2018-05-28:
Well checking against the local thresh is done in line 170.
`if (current_height <= (prev_height + height_threshold) && current_height >= (prev_height - height_threshold))`

So for me it made sense that local height threshold is always compared with the prev_height and  global height threshold is always used with sensor height.

closed 2018-06-04 by amc-nu

kargarisaac, 2018-10-27:
What is the ray ground filter and other filters? I didn't find any document or paper.