autowarefoundation/autoware #1301: Build error docker/generic

andoh104, 2018-05-28:
# Bug
## Expected Behavior

No build error

## Actual Behavior

```
-- +++ processing catkin package: 'amathutils_lib'
-- ==> add_subdirectory(common/libs/amathutils_lib)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'as'
-- ==> add_subdirectory(actuation/vehicles/packages/as)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'module_comm_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
-- Configuring incomplete, errors occurred!
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
####
#### Running command: "cmake /home/autoware/Autoware/ros/src -DCATKIN_DEVEL_PREFIX=/home/autoware/Autoware/ros/devel -DCMAKE_INSTALL_PREFIX=/home/autoware/Autoware/ros/install -G Unix Makefiles" in "/home/autoware/Autoware/ros/build"
####
The command '/bin/sh -c /bin/bash -c 'source /opt/ros/kinetic/setup.bash; cd /home/$USERNAME/Autoware/ros/src; catkin_init_workspace; cd ../; ./catkin_make_release'' returned a non-zero code: 1
```

## Steps to Reproduce the Problem

  1.--no-cache=true
`~/Autoware/docker/generic$ sh build.sh indigo
`

## Specifications

  - Ubuntu version:16.04
  - ROS version:kinetic
  - Autoware branch:master

amc-nu, 2018-05-28:
The sub modules are missing.
Please follow the how to build section in the readme.

andoh104, 2018-05-29:
@amc-nu 
Thank you for your answer.
Don't you mean this README ?
https://github.com/CPFL/Autoware/blob/master/docker/generic/README.md

amc-nu, 2018-05-29:
Yes , exactly that one

andoh104, 2018-05-29:
I think I have already  installed 
- NVIDIA drivers, 
- Docker
- nvidia-docker 

And I was able to build in the same environment 4 weeks ago.

andoh104, 2018-06-02:
@amc-nu

In docker environment, 1.6.3 can be built ,but master and 1.7.0 can't be built .

```
[100%] Linking CXX executable /home/autoware/Autoware/ros/devel/lib/lidar_tracker/euclidean_cluster
[100%] Linking CXX executable /home/autoware/Autoware/ros/devel/lib/lidar_tracker/kf_lidar_track
[100%] Linking CXX executable /home/autoware/Autoware/ros/devel/lib/op_local_planner/op_trajectory_generator
[100%] Built target euclidean_cluster
[100%] Built target op_trajectory_generator
[100%] Built target kf_lidar_track
autoware@A-UX331UN:~/Autoware/ros$ git branch
* (HEAD detached at 1.6.3)
  master
autoware@A-UX331UN:~/Autoware/ros$ 
```

```
-- Could not find the required component 'module_comm_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/autoware/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
autoware@A-UX331UN:~/Autoware/ros$ git branch
* (HEAD detached at 1.7.0)
  master
autoware@A-UX331UN:~/Autoware/ros$ 
```

amc-nu, 2018-06-02:
@andoh104 checking the docker script there’s something missing in line 100

https://github.com/CPFL/Autoware/blob/master/docker/generic/Dockerfile.indigo#L100

Need to be changed for
git clone https://github.com/CPFL/Autoware.git --recurse-submodules

Can you please test?

@messi49 can you please create a PR to both develop and master to fix this?

amc-nu, 2018-06-02:
@yk-fujii can you please check if the nvidia docker build script? Thanks

andoh104, 2018-06-02:
I tested some revisions, and found this commit cause the docker build error.
https://github.com/CPFL/Autoware/commit/676367433dd27ddef9764b44afb97f58a728ea36

castleguarders, 2018-06-12:
Facing the same issue.

lijecaru, 2018-06-12:
@andoh104  I get the exact same error when building, were you able to find the solution? 

@amc-nu I added the lines to test if something would happen, but I only get more errors when building.

andoh104, 2018-06-12:
@cirpue49  Your commit 6763674 cause this docker build error . Can you find the cause ?

kitsukawa, 2018-06-13:
Try this in Autoware directory before compile.
`$ git submodule update --init`

andoh104, 2018-06-14:
@kitsukawa  Thank you for your information.
And I found that it's already fixed in https://github.com/CPFL/Autoware/commit/71fb46e521ca68c047139d2ea763bf96a6a9b6c6 .
I can  build successfully on ｔｈｅ current master branch.

closed 2018-06-14 by andoh104