autowarefoundation/autoware #1302: Obstacle_avoid

Mor-harry, 2018-05-29:
# Bug
  - Ubuntu version:14.04&16.04
  - ROS version:indigo&kinetic
  - Autoware branch:develop

1.Using the Obstace_avoid node and opening the Avoidance option, it is impossible to avoid obstacles in the test of a real car and can only stop before an obstacle.
2.I used the following nodes：
   lane_rule;
   lane_stop;
   lane_select;
   obstace_avoid;
   velocity_set;
   pure_suit;
   euclidean_cluster;
   points2costmap;
   grid_map_filiter;
   I can get  /real_cost_map and send it to grid_map,but it's still parked in front of the barrier, not avoiding it.

javcursor, 2019-02-26:
Hi @Mor-harry,
I'm in a similar situation, did you get a solution?

Thanks

gbiggs, 2019-02-27:
@javcursor Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

closed 2019-02-27 by gbiggs

marcusvinicius178, 2021-09-03:
Hi @gbiggs I am facing this harsh issue https://github.com/Autoware-AI/autoware.ai/issues/2387 for a Long time.
I have tried a lot off code combination/configuration, joined related nodes and  launch files, modified the planners parameters, checkbox...

However I still was not able to solve this alone, the car does not avoid the obstacle anyway. I am sure you can give me some good advice to find a solution. I have followed your comments on git, slash, ROS answers, etc and I believe you can help, you have a solid knowledge about Autoware modules. Could you assist me on this regard please?