autowarefoundation/autoware #1303: "./run" in genetic x86 docker gets errors:"Failed to run /home/autoware/Autoware/ros/run_proc_manager as user root""

Patrick-Woo, 2018-05-30:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
I build the genetic x86 docker followed the wiki install docker Readme. after build the docker successfully, I enter the docker and press ./run.  it occurs a error message"Failed to run /home/autoware/Autoware/ros/run_proc_manager as user root". 

also in one of the green terminals, I can see another error message "(runtime_manager_dialog.py:174): Gdk-CRITICAL **: IA__gdk_color_free: assertion 'color != NULL' failed"
[image]

## Steps to Reproduce the Problem

  1.build the x86 docker 
  2.Autoware Docker Setup with case 2 method:   sudo sh build.sh kinetic
  3.sudo sh run.sh kinetic
  4. cd Autoware/ros
  5.    ./run
  6. get errors mentioned above

## Specifications

  - Ubuntu version:ubuntu16.04.1 LTS
  - ROS version: kentic
  - Autoware branch: master

lijecaru, 2018-06-12:
@Patrick-Woo did you find a solution? I'm experiencing a similar problem with my installation.

HiwasawaWorks, 2018-06-25:
@Patrick-Woo 
@lijecaru 
Hi, I have same error too.
Did you solve ?

lijecaru, 2018-06-27:
Hi @Patrick-Woo , I reinstalled everything. It turned out that I didn't have all the pre requisites (I didn't have all the drivers from Nvidia installed since it was a fresh Ubuntu install, I also didn't have Qt). After that, I erased  all the previous information from autoware, uninstalled the docker and followed all the steps. Then it worked.

However, before that I attempted using autoware without installing the docker, and it worked as well (https://github.com/CPFL/Autoware/wiki/Source-Build). But of course, the docker is recommended, that's why I uninstalled all the previous stuff and reinstalled the docker.

faint5, 2018-07-05:
Hi @lijecaru
i have same proble.
Can you tell me what shoud i install the pre requisites in a fresh Ubuntu?
Please help me!!

lijecaru, 2018-07-06:
Hi @faint5 , I just followed the list of prerequisites. I installed OpenCV (the latest version) and Qt (the latest version as well). You can check if you have the Nvidia driver in System Settings/Software&Updates/Additional Drivers. If you don't find it, then you can download the Nvidia driver from the source (I think I followed the instructions here http://www.linuxandubuntu.com/home/how-to-install-latest-nvidia-drivers-in-linux but instead of the 370 version, you can choose what you need, I think I have the 390).

You can also try to use autoware without the docker if it's urgent. Please let me know if it works.

faint5, 2018-07-18:
I solved my problem by installing autoware source. Thank you!!!!

filiperinaldi, 2019-02-05:
This issue is a duplicate of #1352. Although this issue is older by a few days, the other issue is more precise with regards the error. I suggest someone (@kfunaoka ?) to close this issue as duplicated.

amc-nu, 2019-03-20:
Thanks @filiperinaldi

datnguyen82, 2019-06-14:
@lijecaru , did you install Qt and OpenCV in the docker file or it also works if the software are installed somewhere else in the computer? thanks!

closed 2019-06-16 by gbiggs