autowarefoundation/autoware #1306: BUG

ShitianyuBIT, 2018-05-31:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'module_comm_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/shitianyu/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/shitianyu/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

## Specifications

  - Ubuntu version:16.04LTS
  - ROS version:ROS kinetic

amc-nu, 2018-05-31:
You're skipping one step. Please follow:

https://github.com/CPFL/Autoware#how-to-build

closed 2018-05-31 by amc-nu

ShitianyuBIT, 2018-06-01:
OK,I have solved my problem ,thanks