autowarefoundation/autoware #1309: The information on waypoints

wsung1, 2018-06-03:
Hi, all,
Now I'm somewhere inbetween the path generation and the path planning. I've got a path generated by /waypoint_clicker. The used vector maps for this are included in sample_moriyama_data. The path generated by Publish Point (actually, Op Flag) on RViz is going along a dim yellow line
[image]
This path is then saved as
[image]
...
Here, only 24 rows are displayed, but actually it includes 474 rows, which means 474 waypoints.
1) how can the number of waypoints be decided?
And I think
 - 1st~3rd columns refer to the coordinate (x, y, z, respectively) of waypoints
 - 4th column represents the direction
 - 5th column denotes the (target) velocity.
I've plotted those
[image]
[image]
At intersection, the value of the direction abruptly changes from about +2 to -2~-3.
2) what is the unit (orientation) of the direction?
Thanks for your interest!

cirpue49, 2018-06-06:
Hi, 

Question1: Please check these codes, [waypoint_clicker.cpp](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/motion/packages/waypoint_maker/nodes/waypoint_clicker/waypoint_clicker.cpp), [lane_planner_vmap.hpp](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/mission/packages/lane_planner/include/lane_planner/lane_planner_vmap.hpp)
The number of points are determined by your clicks and vectormap info.

Question2: Unit for orientation(yaw) is radian.

wsung1, 2018-06-06:
@cirpue49, thanks for your reply :)
If possible, would you please a bit more elucidate your answer 2)?
Yes, the unit is radian (0~2*pi) but why does it change from plus to minus while passing by the intersection in the middle?
I've just searched for the possible orientations of vehicle heading (yaw) but there are many
http://street.umn.edu/VehControl/javahelp/HTML/Definition_of_Vehicle_Heading_and_Steeing_Angle.htm
https://stackoverflow.com/questions/17574424/how-to-use-atan2-in-combination-with-other-radian-angle-systems
http://docs.polysync.io/articles/tutorials-and-examples/tutorials/vehicle-sensor-alignment/
...
Would you tell me the orientation of vehicle heading (yaw) defined in Autoware?
Thank you, @cirpue49!

cirpue49, 2018-06-06:
Please check the code.
[lane_planner_vmap.cpp#405](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/mission/packages/lane_planner/lib/lane_planner/lane_planner_vmap.cpp#L405)

It uses atan2.

wsung1, 2018-06-07:
@cirpue49, all right, thank you! :)

closed 2018-06-07 by cirpue49