autowarefoundation/autoware #1314: Using rqt(_plot) with Autoware

wsung1, 2018-06-04:
Hi, Autoware,
Has anyone run rqt(_plot) with Autoware? As shown below, path-following is in progress, generating /twist_cmd. This topic can be monitored using Topics\Echo in runtime manager and also rostopic list.
[image]
[image]
Then, I'd like to plot this topic through rqt_plot but typing just rqt_plot (rqt_plot twist_cmd/angular/z) went wrong as shown below.
[image]
**1) What causes this problem and how can I fix it?**
and
**2) How can I use RQT (button) next to RViz on runtime manager?**
[image]
Your help will be greatly appreciated. Thanks in advance!

cirpue49, 2018-06-06:
Hi, 
First things first, you should subscribe to `/twist_cmd/twist/angular/z` instead of `twist_cmd/angular/z`.

Question1, The message says that autoware::msg is not recognized. This would be caused by autoware is not correctly built or PATH is not loaded(e.g. source devel/setup.bash)

Question2, Check the plugin tab. 
[image]

wsung1, 2018-06-07:
Thanks for your reply, @cirpue49.
When it comes only to 1), assuming my Autoware is correctly built, I need to check whether a PATH is properly loaded.
The PATH is loaded by running .bashrc whenever ROS starts, which is done by
cd Autoware/ros/src
source /opt/ros/kinetic/setup.bash
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
source ~/.bashrc
, which is done when installing Autoware.
Do I have to specifically set a different PATH for executing rqt_plot?
Sorry for bothering you @cirpue49 but please consider this issue with me.

cirpue49, 2018-06-07:
Hi, 

You can check it from section3 in this [wiki](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment)
Please make sure your wokspace is overrayed in `ROS_PACKAGE_PATH`.

wsung1, 2018-06-07:
As advised, I've got my workspace overlaid in ROS_PACKAGE_PATH by doing
export ROS_PACKAGE_PATH=/home/futuremobilitylab01/Autoware/ros/src:$ROS_PACKAGE_PATH
[image]
However, as shown above, the same error took place even though sudo rosdep init and then rosdep update were done.

cirpue49, 2018-06-08:
Please read the wiki.
could you run `source devel/setup.bash` instead of running `export ROS_PACKAGE_PATH=/home/futuremobilitylab01/Autoware/ros/src:$ROS_PACKAGE_PATH`?

closed 2018-06-13 by cirpue49