autowarefoundation/autoware #1317: The vehicle never start in simulating the local planning by using OpenPlanner

wsung1, 2018-06-05:
Hi, Autoware!
In order to run OpenPlanner, I've been following https://www.youtube.com/watch?v=FKM8v79X3_s (I'd appreciate @hatem-darweesh!).
As I understand, this footage introduces two different ways of the local planning simulation using
 1) ff_waypoint_follower [05:01~09:03]
 2) waypoint_pursuit [09:04~10:45]
I had no problem in performing the simulation with 1); however, not with 2).
As shown below, the vehicle (model) never hit the road after checking pure_pursuit on.
trash:///do%20not%20go.png
In the footage, at the moment 2) starts [09:04],
 - way_planner is checked on
 - (blue-colored) global path is active (as displayed in RViz)
 - vel_pose_connect is checked on
 - (dp_planner is checked off)
With this condition, I've just followed what @hatem-darweesh did in order of
 1) wf_simulator
 2) twist_filter
 3) self-localize using 2D Pose Estimate in RViz
 4) dp_planner
 5) (finally,) pure_pursuit
But the vehicle (model) doesn't go.
**What can be the culprit of this problem?**
Thanks for reading, @hatem-darweesh :)

hatem-darweesh, 2018-06-07:
1- Setup -> TF , Vehicle Model 
2- Map -> Point Cloud map , Vector Map, TF 
3- Computing: 
  a) vel_pose_connect -> check "Simulation Mode" 
  b) twist_filter and tf_simulator, then select start pose from RVIZ using "2D Pose Estimate"
  c) way_planner (after set map path), select goal pose from RVIZ using "2D Nav Goal" 
  d) dp_planner (after set map path) 
  e) pure_pursuit -> select Waypoint 

Car should move.

Regards,

wsung1, 2018-06-07:
Thanks for your reply, @hatem-darweesh!
By following the sequence you gave, the vehicle model is now moving!

hatem-darweesh, 2018-06-07:
Great, 
Can you share with us what was wrong with your steps, so I can consider in the tutorial. 

Regards,

wsung1, 2018-06-07:
I think I was wrong in setting start and goal positions on RViz with way_planner. As demonstrated, this step should be done after wf_simulator and before dp_planner.
I'd appreciate your support, @hatem-darweesh!

hatem-darweesh, 2018-06-08:
A lot of these issues is solved with the new OpenPlanner 1.5 if you have spare time give it a try feature/OpenPlanner and tutorial video is on the OpenPlanner youtube channel.

closed 2018-06-08 by hatem-darweesh

hatem-darweesh, 2019-01-20:
@DmitryRedin OpenPlanner uses map coordinates , and to use 2D Nav Goal it should be in world coordinates because the global planner convert it to map coordinate