autowarefoundation/autoware #1319: Not building up map when .bag file created in Gazebo is put.

yunjeongkim, 2018-06-07:
Hi, I tried to building a map using Autoware ndt_mapping with an input of .bag file created in Gazebo. 
I expected the map is build up but the map does not be built and it only shows just the first scene of point cloud data. 

I used  sample_moriyama bag file as an input, it works very well. 

I recorded PointCloud2 data published in /points_raw topics in Gazebo using Runtime Manager. 

  - Ubuntu version: 14.04
  - ROS version: indigo
  - Autoware branch: 1.7.0

Thanks in advance.

Nicholas-Mitchell, 2018-06-15:
I am currently having the same issue. I recorded my own rosbag file, which contains only one topic holding PointCloud2 messages.

[ I followed this tutorial ](https://youtu.be/ss6Blrz23h8) and the only thing that I'm not sure about the part where they set up the "Baselink to Localizer" in the Setup tab of Autoware's interface - 35 seconds into that video.

If I only have that single topic of PointCloud2 messages, is something missing, which `ndp_mapping` requires? I found the documentation for it, where there is a flow chart showing only point cloud data as an input.

libaiaibiancheng, 2018-07-05:
Hi @yunjeongkim 
I am currently having the same issue. I saw you have a  solution, but i don't know how to do it. Therefore, I hope that you can send your points map to me. In fact, I only need  the points map to simulate, this is the fastest way at present.  
My email address is wonghoa@126.com. 
Thanks in advance.

yunjeongkim, 2018-07-05:
I solved this problem. 
I used libgazebo_ros_velodyne_laser.so plugin which I made by compiling the source in https://bitbucket.org/DataspeedInc/velodyne_simulator, and added gazebo plugin path in .bashrc file so that gazebo can find the .so file.
Also, It is important to comment <collision> tab of velodyne link in catvehicle.xacro.
If you do that, you can obtain velodyne point cloud.

gzblue07, 2018-10-23:
Hi @yunjeongkim ,
I tried to use libgazebo_ros_velodyne_laser.so plugin,but NDT_mapping still failed,can you helped me to build the map?  or let me know what step I ignore,so that I can not build map well,thanks!

gzblue07, 2018-10-23:
Hi @yunjeongkim 
I used libgazebo_ros_velodyne_laser.so,and I can display the points_raw,but I still couldn't build the map,what's wrong with my setting?thank you!

yukkysaito, 2019-02-16:
Old gazebo will be replace the new Gazebo.
This issue will be closed.
#1930

closed 2019-02-22 by yukkysaito