autowarefoundation/autoware #1320: Use SICK LMS511 to mapping  with Autoware

vehicularkech, 2018-06-07:
Hi,eveyone:
    I have downloaded a rosbag of LMS511 and  also have converted Laserscan messages to PointCloud2,but when i use the the pointcloud2 data to generate pcd file by ndt_mapping, error happens:
**"
**[pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!**

"** 
Could you suggest me how to resolve the issue ? thanks.
[image]

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version: ubuntu 16.04
  - ROS version: kinetic
  - Autoware branch: master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

kitsukawa, 2018-06-07:
ndt_mapping only support 3D scan and does not work with 2D scan.

vehicularkech, 2018-06-07:
Thanks for your replay.

closed 2018-07-30 by amc-nu