autowarefoundation/autoware #1322: catkin make fail in lattice_planner

arashroomi, 2018-06-08:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
catkin make of lattice_planner

## Actual Behavior
 
Failed and produce this error:

/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgetrf_'
/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgetri_'
/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgemv_'
collect2: error: ld returned 1 exit status
computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_twist_convert.dir/build.make:321: recipe for target '/home/carpc/Autoware-master/ros/devel/lib/lattice_planner/lattice_twist_convert' failed
make[2]: *** [/home/carpc/Autoware-master/ros/devel/lib/lattice_planner/lattice_twist_convert] Error 1
CMakeFiles/Makefile2:39570: recipe for target 'computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_twist_convert.dir/all' failed
make[1]: *** [computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_twist_convert.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Building CXX object computing/planning/motion/packages/op_local_planner/CMakeFiles/op_trajectory_generator.dir/nodes/op_trajectory_generator/op_trajectory_generator_core.cpp.o
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/points_preprocessor/points_concat_filter
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/lattice_planner/lattice_trajectory_gen
/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgetrf_'
/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgetri_'
/home/carpc/Autoware-master/ros/devel/lib/liblibtraj_gen.so: undefined reference to `wrapper_dgemv_'
collect2: error: ld returned 1 exit status
computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_trajectory_gen.dir/build.make:321: recipe for target '/home/carpc/Autoware-master/ros/devel/lib/lattice_planner/lattice_trajectory_gen' failed
make[2]: *** [/home/carpc/Autoware-master/ros/devel/lib/lattice_planner/lattice_trajectory_gen] Error 1
CMakeFiles/Makefile2:39423: recipe for target 'computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_trajectory_gen.dir/all' failed
make[1]: *** [computing/planning/motion/packages/lattice_planner/CMakeFiles/lattice_trajectory_gen.dir/all] Error 2
[100%] Built target points_concat_filter
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/object_map/wayarea2grid
[100%] Built target ray_ground_filter_lib
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/lidar_kf_contour_track/lidar_kf_contour_track
[100%] Built target wayarea2grid
[100%] Built target lidar_kf_contour_track
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/points_preprocessor/cloud_transformer
[100%] Linking CXX executable /home/carpc/Autoware-master/ros/devel/lib/op_local_planner/op_trajectory_generator
[100%] Built target op_trajectory_generator
[100%] Built target cloud_transformer
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

## Steps to Reproduce the Problem

  1. ./catkin_make_release

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: master

kfunaoka, 2018-07-18:
@arashroomi Cannot reproduce. Which commit are you using?

```
$ git clone git@github.com:CPFL/Autoware.git Autoware_Issue1322
Cloning into 'Autoware_Issue1322'...
remote: Counting objects: 57183, done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 57183 (delta 5), reused 10 (delta 2), pack-reused 57161
Receiving objects: 100% (57183/57183), 257.26 MiB | 5.55 MiB/s, done.
Resolving deltas: 100% (31886/31886), done.
Checking connectivity... done.
$ cd Autoware_Issue1322/ros
$ git log -n1
commit 470b04b95eee435d2bb529d978d4c91bf57f34b0
Merge: b63a0ef 01a9199
Author: Kenji Funaoka <kfunaoka@gmail.com>
Date:   Wed Jul 18 11:40:21 2018 +0900

    Merge pull request #1378 from kfunaoka/hotfix/revert_readme_slack_uri
$ ./catkin_make_release
...
[100%] Built target road_occupancy_processor
[100%] Built target grid_map_filter
[100%] Linking CXX executable /home/funaoka/Autoware_Issue1322/ros/devel/lib/points_preprocessor/ray_ground_filter
[100%] Built target ray_ground_filter
```

https://github.com/CPFL/Autoware/commits/master shows which commit passes CI.

kfunaoka, 2019-05-16:
We can go to [ROS Answers](https://answers.ros.org/questions/) if some problems remain.

closed 2019-05-16 by kfunaoka