autowarefoundation/autoware #1323: How to use lidar_svm_detect package ?

jediofgever, 2018-06-09:
Hi , 

Thanks for amazing work and support of this free software. 
I would like to use your lidar_svm_detect package under perception module. However in code I see that you have given a path to svm.model but I cant see the path and model neither.

How can I make use of that package? , Do I need that svm.model to be able to use this package ? 

Thank you for your `help.`

jediofgever, 2018-06-11:
Please assign someone from lidar_detect package to help me with this issue

cirpue49, 2018-06-13:
Currently, lidar_svm_detect package does not have any function.
Your contribution to lidar_svm_detect or another method for lidar classification would be greatly appreciated.