autowarefoundation/autoware #1325: What is the differece between the scoring system of your ndt_gpu method and pcl's?

DriftingTimmy, 2018-06-11:
When I distracted ndt_gpu method in Autoware to use it in my project. I found that the final score of ndt_gpu method is quite different with the pcl's.
While the pcl's method is getting the match score to 0.1 to 1 point. The ndt_gpu always get a much higer score to 6 to 10 points.
So can u tell me the differece between the scoring system of your ndt_gpu method and pcl's?

anhnv3991, 2018-06-12:
@DriftingTimmy Are you talking about the Fitness Score?
Fitness Score is the average distance (in meters) between each scanned point (points produced by the car's scanner) and its nearest point on the map. It is used to avoid the cases of miss-aligned (such as the location of the car determined by matching process is 100 meters far from its actual location). In such cases, the initial guess of the ndt_matching will be reset.
Fitness score depends on the searching method, which is used to search for the nearest neighbor of each scanned point (such as kd-tree, octree, etc). I used octree for my implementation, but initially it is very slow, so I had to adjust it a bit. As a result, it becomes faster, but less accurate. That's why you see the fitness score of gpu is much higher than pcl's.
I am working on a more accurate searching method, but it is still work-in-progress.

DriftingTimmy, 2018-06-12:
@anhnv3991  Thanks very much! I think u offered a very reasonable explanation. But how can I adjust the threshold to suit the gpu's method to make sure that the error is acceptable while the score is not that small as pcl's? And what do u think can make the matching process more precise? 
And can u give me some suggestions about how to learn cuda programming to rewrite the code of some algorithms if we want to accelerate it?

anhnv3991, 2018-06-13:
@DriftingTimmy 
If you can implement a nearest neighbor search for point clouds in the gpu, the matching process will be more precise. That's what I am trying to do now using octree.
As for learning cuda programming, it's hard for me to give you a suggestion about how to do it because I don't know about your experience about cuda and nvidia gpu (e.g. are you familiar with basic concepts of cuda, have you ever written cuda code, etc).
If you are at the beginner level, then I suggest you try the "cuda by example". To know more about the nvidia gpu, you should try cuda handbook or cuda programming guide of NVIDIA (I prefer the cuda handbook). At higher level, parallelforall is an useful resource for learning. You can find many tips to make your program faster there.
I hope those can help.

DriftingTimmy, 2018-06-13:
@anhnv3991
Thanks for your suggestions! Those tutorial guiding books will help me a lot!
Here are some other questions about the gpu programming.
When we try to handle the huge amount of points' data, I noticed that in your code u use gpu computing each point to replace the iterative process. So do u think we can replace all iterative part with gpu programming?

anhnv3991, 2018-06-14:
@DriftingTimmy 
Not really. If iterations depend on each other, simply moving them to the GPU is not easy. Either it does not work or it just have poor performance. Sometimes, executions in each iteration is also a problem because they are complicated and cannot be copy-pasted to the GPU (because each GPU thread has very limited computing resources). In such cases, you will have to find some way to reduce the computing resources in each iteration or find another parallel algorithm that fits the GPU. That depends a lot on your problem.