autowarefoundation/autoware #1329: Record points map from velodyne data of gazebo

gnykumar, 2018-06-12:
# Bug
## Expected Behavior
Record points map from velodyne data of gazebo

## Actual Behavior
Unable to create points map, only one instance is updating not further.

## Steps to Reproduce the Problem

  1.Run gazebo
  2. Run respective tf
  3. Run ndt_mapping
  4. Run rviz for ndt_mapping.rviz
  5. Run /twist_cmd give cmd to gazebo to move
  6. Check map updating  in Rviz

## Specifications

  - Ubuntu version: 14.04
  - ROS version: indigo
  - Autoware branch:

yunjeongkim, 2018-06-15:
I had the same problem. If you use /points_raw topic which is converted from /catvehicle/points_raw, the map is not built because ring field in PointCloud2 data is missing. 
One solution is to produce PointCloud2 data directly using  libgazebo_ros_velodyne_laser.so plugin. 
This will fix your problem.

gnykumar, 2018-06-16:
Thank you, I will try the above and update back.

gnykumar, 2018-06-19:
<plugin name="laser_controller" filename="libgazebo_ros_block_laser.so">
			      <frameName>velodyne</frameName>
			      <hokuyoMinIntensity>101</hokuyoMinIntensity>
			      <robotNamespace>/$(arg roboname)</robotNamespace>
			      <topicName>points_raw</topicName>
			      <gaussianNoise>0.02</gaussianNoise>
			      <updateRate>30</updateRate>
                 </plugin>

is the plugin used in my workspace, if I modify from current to libgazebo_ros_velodyne_laser.so I am not getting Velodyne out. But with current, I am getting output but can't record 3D pcd map file

yunjeongkim, 2018-06-19:
I used libgazebo_ros_velodyne_laser.so plugin which I made compiling the source in https://bitbucket.org/DataspeedInc/velodyne_simulator, and added gazebo plugin path in .bashrc file so that gazebo can find the .so file that I made. 
Also, It is important to commnet <collision> tab of velodyne link in catvehicle.xacro. 
If you do that, you can obtain velodyne point cloud.

gnykumar, 2018-06-20:
Does gazebo version matters?

yunjeongkim, 2018-07-05:
I use Gazebo2 with ROS indigo. I don't know whether it is because of the version matter or not. 

I hope that you solve this problem.

gnykumar, 2018-07-06:
Thank you. I am also using gazebo2 only. I am still trying on same.  I will update back once I am able to get it. If not i will update respective issue.

Thank you again. :)

gzblue07, 2018-10-23:
@gnykumar hi,have you solved this problem?tks!

gnykumar, 2018-11-26:
@gzblue07 Hi, No.

yukkysaito, 2019-02-16:
Old gazebo will be replace the new Gazebo.
This issue will be closed.
#1930

closed 2019-02-20 by gnykumar