autowarefoundation/autoware #1331: ResourceNotFound: ndt_localize in build from master

castleguarders, 2018-06-13:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
ndt_localizer fails to load when trying to run the sample_moriyama. Build process works fine and finishes building everything.

## Expected Behavior
localization should work when running the sample.

## Actual Behavior
Cannot find the resource..ResourceNotFound: ndt_localize

## Steps to Reproduce the Problem
From a build from git master
  1. Run sample simulation
  2. Pick my_map.launch and set
  3. Pick my_localization.launch and set 

Console logs the error messages.
ResourceNotFound: ndt_localize
source_path_to_packages=source_path_to_packages)
File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
full_path = _get_executable_path(rp.get_path(args[0]), path)
File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: ndt_localizer
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/castleguard/Autoware/ros/src
ROS path [2]=/home/castleguard/projects/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src
ROS path [3]=/opt/ros/kinetic/share

## Specifications
  - Ubuntu version: 16.04
  - ROS version: kinetic 
  - Autoware branch: master

castleguarders, 2018-06-13:
Another ResourceNotFound error when trying to load my_sensing.launch from the sample. This is from a fresh build off the head, with all submodules checked out. The build itself goes flawless, runtime has issues.

raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: pointgrey

hadiTab, 2018-06-15:
Package names have been changed in release 1.7 but it seems like the demo launch files have not been updated.
You can fix this by editing my_localization.launch and replace ndt_localizer with lidar_localizer and remove the line below it which uses the use_openmp argument.

hadiTab, 2018-06-15:
Yes, it is now called autoware_pointgrey_drivers.

castleguarders, 2018-06-15:
Thanks, that gets a bit further. However, looks like grasshopper has been renamed as well?

ERROR: cannot launch node of type [autoware_pointgrey_drivers/grasshopper3_camera]: can't locate node [grasshopper3_camera] in package [autoware_pointgrey_drivers]

hadiTab, 2018-06-15:
It's been renamed to grasshopper3.launch

castleguarders, 2018-06-15:
hmm, there are references in cmake files to grasshopper3_camera and ladybug_camera.. so this one seems a bit more involved.

hadiTab, 2018-06-15:
Oh, sorry my bad, the name hasn't changed. It is still grasshopper3_camera and ladybug_camera. Have you recompiled since pulling the latest source?

castleguarders, 2018-06-15:
Yes, a couple of times. Ended up changing cmake files and seeing if that works. Based on your comment, it seems like that's the wrong track to take.

castleguarders, 2018-06-15:
No luck with that as expected.

castleguarders, 2018-06-20:
The issue turned out to be that the grasshopper3 node wasn't being built due to missing dependencies. Had to get the flycapture SDK among other sub dependencies. 

It seems excessive to need this for running the sample..

130s, 2018-06-25:
> The issue turned out to be that the grasshopper3 node wasn't being built due to missing dependencies. Had to get the flycapture SDK among other sub dependencies.

Is that dependency not defined in `package.xml` anywhere? If it is, the `ResourceNotFound` error may or may not be because of a few bugs in `rospkg` I've been reporting (e.g. https://github.com/ros-infrastructure/rospkg/issues/142 and/or https://github.com/ros-infrastructure/rospkg/issues/131).

andoh104, 2019-03-27:
How about this issue ? The same problem occurs in my environment.

closed 2020-05-29 by gbiggs