autowarefoundation/autoware #1336: Build error on ./catkin_make_release

gulshan216, 2018-06-18:
# Bug
Scanning dependencies of target rosinterface
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/get_boxes.cpp.o
[ 66%] Building CXX object util/packages/RobotSDK/rosinterface/CMakeFiles/rosinterface.dir/ROSInterface/rosinterface.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/load_model.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/multithreading.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/nms.cpp.o
Scanning dependencies of target autoware_rviz_plugins
[ 66%] Building CXX object util/packages/autoware_rviz_plugins/CMakeFiles/autoware_rviz_plugins.dir/src/state_panel/state_panel.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/obstacle_detection.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/resize_GPU.cpp.o
[ 66%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/tracking.cpp.o
[ 66%] Building CXX object util/packages/RobotSDK/rosinterface/CMakeFiles/rosinterface.dir/include/rosinterface/moc_rosinterface.cpp.o
[ 66%] Linking CXX shared library /home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/liblibdpm_ttic.so
/usr/bin/ld: cannot find -lcuda
collect2: error: ld returned 1 exit status
computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/build.make:747: recipe for target '/home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/liblibdpm_ttic.so' failed
make[2]: *** [/home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/liblibdpm_ttic.so] Error 1
CMakeFiles/Makefile2:15243: recipe for target 'computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/all' failed
make[1]: *** [computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 66%] Building CXX object util/packages/RobotSDK/rosinterface/CMakeFiles/rosinterface.dir/rosinterface_autogen/mocs_compilation.cpp.o
[ 66%] Building CXX object util/packages/autoware_rviz_plugins/CMakeFiles/autoware_rviz_plugins.dir/src/sim_object/autoware_flag_tool.cpp.o
[ 66%] Building CXX object util/packages/autoware_rviz_plugins/CMakeFiles/autoware_rviz_plugins.dir/autoware_rviz_plugins_autogen/mocs_compilation.cpp.o
[ 66%] Linking CXX executable /home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/multi_lidar_calibrator/multi_lidar_calibrator
[ 66%] Built target multi_lidar_calibrator
[ 66%] Linking CXX shared library /home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/librosinterface.so
[ 66%] Built target rosinterface
[ 66%] Linking CXX shared library /home/pl/Documents/my_work/Autoware/Autoware/ros/devel/lib/libautoware_rviz_plugins.so
[ 66%] Built target autoware_rviz_plugins
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

## Expected Behavior
The build should complete successfully according to the instructions

## Actual Behavior
The build fails with 'cannot find -lcuda', even though the requirements mentions CUDA is optional

## Steps to Reproduce the Problem

$ cd ~/Autoware/ros/src
$ catkin_init_workspace
$ cd ../
$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
$ ./catkin_make_release

## Specifications

  - Ubuntu version: 16.04 LTS
  - ROS version: Indigo
  - Autoware branch: Latest Development Branch

kfunaoka, 2018-07-13:
Would you check CUDA_FOUND in libdpm_ttic?

```diff
diff --git a/ros/src/computing/perception/detection/vision_detector/libs/dpm_tti
index 57d9f347..9cd9118f 100644
--- a/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/CMake
+++ b/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/CMake
@@ -23,6 +23,7 @@ set(CMAKE_C_FLAGS "-O2 -g -Wall -Wno-unused-result ${CMAKE_C_F
 set(CMAKE_CXX_FLAGS "-std=c++11 -O2 -g -Wall -Wno-unused-result ${CMAKE_CXX_FLA
 
 if(CUDA_FOUND)
+    message("CUDA_FOUND in libdpm_ttic")
     include_directories(
             include
             ${catkin_INCLUDE_DIRS}
```

kfunaoka, 2018-07-13:
@gulshan216

amc-nu, 2018-07-30:
@gulshan216 seems you have an incorrect CUDA/NVidia Drivers setup.

closed 2018-07-30 by amc-nu