autowarefoundation/autoware #1342: Problem when adding simulated traffic in Rviz

yunjeongkim, 2018-06-21:
I have followed the OpenPlanner tutorial (https://www.youtube.com/watch?v=FKM8v79X3_s&t=63s)

but I found the message such as 

`Performance Alert: Can't Find Lane Information in Global Path, Searching the Map :( `

when I launched OpenPlannerSimulator2.launch and set the initial pose in Rviz. 

I modified OpenPlannerSimulator2.launch so that "mapFileName" 's default value indicates vector map folder's location.  

[image]

Thanks in advance. 

## Specifications

  - Ubuntu version: 14.04
  - ROS version: Indigo
  - Autoware branch: 1.7

yunjeongkim, 2018-06-25:
Hi, I missed to tell you that the pink simulation car appeared in your tutorial does not appear in my case showing the messages mentioned above. 

What can I do to fix this problem? 

Thanks in advance.

hatem-darweesh, 2018-06-25:
if you want to use simulated cars, I recommend using branch feature/OpenPlanner 
and follow tutorial 
https://youtube.com/watch?v=BS5nLtBsXPE&list=PLVAImlqqGbr5G1EjscMom6_JcwjxCMX2g&t=0s&index=17

hatem-darweesh, 2018-06-28:
Dear @hadiTab  Maybe new feature is added, but should not affect the basic operation. 
the video I referenced above for OpenPlanner 1.5 which represent the branch feature/OpenPlanner

closed 2020-05-29 by gbiggs