autowarefoundation/autoware #1349: Lidar Mapping - Flat Area, Multiple Layer Map

cah-icuro, 2018-06-25:
# Bug

## Expected Behavior
Map generation should generate an accurate map when revisiting same location multiple times during lidar scanning.

## Actual Behavior
When generating a map from a lidar scan of a flat area, one will often end up with a multiple-layer map if the lidar scan visits the same area more than once.  An example using data from the rosbag store is [this Odaiba rosbag](https://rosbag.tier4.jp/rosbag_details/?id=203).  Some screenshots below show the problem, all lidar scans were gathered on single-level approximately flat areas.

Odaiba pointcloud, generated without IMU data:

[image]

[image]

Odaiba pointcloud, generated with IMU data (the layers are closer together here):

[image]

[image]

Pointcloud from self-gathered lidar scan, generated without IMU data:

[image]

[image]

## Steps to Reproduce the Problem

  1. Download [Odaiba rosbag](https://rosbag.tier4.jp/rosbag_details/?id=203).
  2. Use Autoware's ndt_mapping node to generate a pointcloud. Mode: CPU, parameters: default values for resolution, step size, etc...

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic
  - Autoware branch: observed in various versions, including `commit e9c9bab2471524c14303f645ca6983d6c9113d3c`

kitsukawa, 2018-06-26:
@cah-icuro Thank you for your report. This problem is also reported in #479 . We are working on this problem and developing loop closure function with graph-based SLAM. But I can not say when it will be released.

jyakaranda, 2019-05-23:
I added odometry based loop-closure detection for ndt_mapping, hope it can help you. [ndt_map with loop-closure detection](https://github.com/jyakaranda/ddl_0725/tree/dev_zh/map_0725/ndt_map)

javcursor, 2019-05-24:
Hi @jyakaranda, thanks for the code. I'm triying it, I've a rosbag with velodyne, gps and imu topics( /points_raw, /imu_raw, /nmea_sentence). I launch the test.launch and the rosbag play
`rosbag play bagfilename.bag --clok /points_raw:=/lslidar_point_cloud /imu_raw:=/imu/data` but I get an error and I don't see anything in Rviz.

```
jcurado@PW7-1-631:~$ roslaunch ndt_map test.launch 
... logging to /home/jcurado/.ros/log/6383fdec-7e10-11e9-9d9e-c8cbb841ba06/roslaunch-PW7-1-631-25628.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PW7-1-631:34607/

SUMMARY
========

PARAMETERS
 * /map_node/ds_history_size: 0.5
 * /map_node/history_fitness_score: 0.3
 * /map_node/history_search_num: 25
 * /map_node/history_search_radius: 10.0
 * /map_node/keyframe_dist: 0.3
 * /map_node/loop_closure_enabled: True
 * /map_node/max_iters: 30
 * /map_node/max_scan_range: 200
 * /map_node/min_scan_range: 24
 * /map_node/ndt_res: 1.0
 * /map_node/step_size: 0.1
 * /map_node/use_imu: True
 * /map_node/use_odom: False
 * /map_node/voxel_leaf_size: 2.1
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /tf_b2l_pitch: 0.0
 * /tf_b2l_roll: 0.0
 * /tf_b2l_x: 2.2
 * /tf_b2l_y: 0.0
 * /tf_b2l_yaw: 0.0
 * /tf_b2l_z: 1.86
 * /use_sim_time: True

NODES
  /
    map_node (ndt_map/map_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [25666]
process[map_node-2]: started with pid [25667]
[ INFO] [1558696591.348841622]: init.
[ INFO] [1558696596.344755913, 1558674739.228557227]: before ds: 12972 points
[ INFO] [1558696596.346531604, 1558674739.228557227]: after ds: 261 points
[ INFO] [1558696596.346579887, 1558674739.228557227]: first laser frame.
-----------------------------------------------------------------
Sequence number: 2; time elapsed: 0.0874244
Number of scan points: 261 points.
map: 261 points.
NDT has converged: 1
Fitness score: 0.0708263
Number of iteration: 0
(x,y,z,roll,pitch,yaw):
(0, 0, 0, 0, -0, 0)
Transformation Matrix:
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
shift: 0
-----------------------------------------------------------------
[map_node-2] process has died [pid 25667, exit code -11, cmd /home/jcurado/Autoware/ros/devel/lib/ndt_map/map_node __name:=map_node __log:=/home/jcurado/.ros/log/6383fdec-7e10-11e9-9d9e-c8cbb841ba06/map_node-2.log].
log file: /home/jcurado/.ros/log/6383fdec-7e10-11e9-9d9e-c8cbb841ba06/map_node-2*.log

```

I need config something more?

Thanks a lot

jyakaranda, 2019-05-25:
> Hi @jyakaranda, thanks for the code. I'm triying it, I've a rosbag with velodyne, gps and imu topics( /points_raw, /imu_raw, /nmea_sentence). I launch the test.launch and the rosbag play
> `rosbag [...]

I've mentioned this issue in [README_en.md](https://github.com/jyakaranda/ddl_0725/blob/dev_zh/map_0725/ndt_map/README_en.md#issues), the package is using odom msg which you didn't provide as initial guess input. Maybe you can just generate odom msg using imu data and relaunch the file. I'll add non-odom support next week also.

farhang760, 2019-11-26:
Hi,  @jyakaranda.
thanks for your code. did you change your code for  non-odom support ?

closed 2020-05-29 by gbiggs