autowarefoundation/autoware #1352: Error copying '/home/root/.Xauthority' to '/tmp/libgksu-NSxrP6': Is a directory

HiwasawaWorks, 2018-06-26:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug

I tried Source Building Autoware following the Users Guide "1.Installation > Docker > Generic X86 Docker".
I entered the docker, Launched the "~/Autoware/ros/Run".

1. An error has occurred on terminal.
   "Error copying '/home/root/.Xauthority' to '/tmp/libgksu-XwkdjK': Is a directory".

2. and popup error "Failed to run /home/autoware/Autoware/ros/run_proc_manager as user root."
   "Unable to copy the user's Xauthorization file."

3. and below message displayed on green terminal
   (runtime_manager_dialog.py:8828): Gdk-CRITICAL **: IA__gdk_color_free: assertion 'color != NULL' failed

## Steps to Reproduce the Problem
  1.Docker Setup
  2.NVIDIA Docker Setup
  3.Autoware Docker Setkup
  4.Executed Case 1

## Specifications
  - Ubuntu version:16.04 LTS
  - ROS version: kinetic
  - Autoware branch: master (as of 2018/6/25)

[image]

JoeyZhu, 2018-07-05:
same exactly problem

zhyfzy, 2018-07-18:
I have solved it by copying $HOME/.Xauthority to /home/root/

lijecaru, 2018-08-17:
Hi @zhyfzy I have the same problem, but your solution didn't work for me. One question, were you able to copy the demo files within the autoware docker? I can't unzip the files within the docker, and I feel it doesn't work because I unpack the files before starting the docker.

zhyfzy, 2018-08-17:
@lijecaru  please check whether you have permissions to create files, as I unpacked these files in the host too.

lijecaru, 2018-08-17:
@zhyfzy how do I check if I have permissions? Do I just unpack the file using `sudo` at the beginning? Could you please post the lines of code that you used?

ChenZiyueyueyue, 2018-08-23:
@zhyfzy I tried your solution but it doesn't work, and I have already change all of the related file permission.

lijecaru, 2018-08-23:
Hi @zhyfzy and @ChenZiyueyueyue. Finally I made it work. After copying the file .Xauthority from /home/myuser to /home/root, I right-click the file and in the permissions tab I selected: Write and Read for all the users (even though I found that it could cause a security breach in future applications). Anyway, autoware ran without problem after that. 

My next problem was that I wasn't able to load the rosbag files from the docker (since the files were unpacked outside the docker). I found that the solution was to unpack those files inside the share_dir folder in home/root, or to create the folder .autoware in home/root and change the run.sh file of the autoware docker. This webpage provided me useful insight to make the demo run: https://ai4sig.org/2018/07/docker-for-autoware/

LukePhairatt, 2018-08-25:
Hi, I had the same problem with .Xauthority error. Running the docker from a root (sudo -i) solved mine.

endland, 2018-10-02:
"sudo" is added in front of "nvidia-docker" in run.sh script for error1, 2. 
But I coundn't find the solution for error 3.

enijmax, 2018-10-02:
I use the following command before execute ./run
export XAUTHORITY=/home/autoware/.Xauthority
than it can be fixed.

hroachewilson, 2018-10-24:
After copying .Xauthority to root, I had a login loop that I couldn't fix. After spending a couple days trying to fix it I had to reinstall the OS. Would rather not mess with root, even though most likely the issue was a typo in one of my commands. Is there an alternative to messing with root directory?

kossumov, 2019-01-22:
None of the above solved the problem for me. Any solutions?

filiperinaldi, 2019-02-05:
The main problem is likely to be that you are starting the container but the .Xauthority files does not exist. When that happens, Docker simply creates a directory with the name .Xauthority and link it with another directory inside the container. This is just how the --volume,-v option works. There are different reasons why you may don't have the .Xauthority file.
You can try to solve this by:
* Stop your containers
* Remove the directory /home/$USER/.Xauthority/ on your host
* Restart your X server on your host (you may just reboot the machine)
* Check if /home/$USER/.Xauthority exists on your host, and make sure it is a *file*
* Now you should be able to start the container using run.sh

filiperinaldi, 2019-02-05:
BTW, it is the "gksudo" utility that checks if .Xauthority is not a file. This tool has been replaced by normal "sudo" on the develop branch which means this error will be gone on the next release (1.11). I believe not having a .Xauthority file still works fine.

dlgydms96, 2019-02-06:
I solved this problem with following steps.
First, edit docker/generic/run.sh in host.
XAUTH=/home/$USER/.Xauthority/.Xauthority
And, after running container, execute Runtime Manager with root permission.

I hope this will help you!

dlgydms96, 2019-02-07:
I solved the problem with following steps.

First, edit Autoware/docker/generic/run.sh.
XAUTH=/home/$USER/.Xauthority/.Xauthority

Second, change permission of .Xauthority
cd /home/root/.Xauthority
sudo chmod 666 .Xauthority

I hope this will help you.

xpharry, 2019-03-20:
If you go with sudo, as below
```
$ sudo ./run.sh
```
you will be led into "/home/root". That is why you can't find the shared files in "/home/user/shared_dir", because you are located in "/home/root/shared_dir". Same thing for the ".Xauthority" file issue.

But without sudo, the issue will be gone. Simply,
```
$ ./run.sh
```

datnguyen82, 2019-06-14:
I'm still struggling with the error: (runtime_manager_dialog.py:8828): Gdk-CRITICAL **: IA__gdk_color_free: assertion 'color != NULL' failed. Anyone who managed to solve this please help!!! Thanks so much!

closed 2020-05-29 by gbiggs