autowarefoundation/autoware #1353: how to load my own pcd map?

StanleyHusai, 2018-06-26:
# Bug
## Expected Behavior
i can load example map data successfully like the video shows (https://www.youtube.com/watch?v=OpvTeTaiXo4). But how to load my own pcd map data? i did step by step as the video shows, but there is no map shown in rviz. maybe need to modify the topic or something? Thanks very much for your answers!

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version: 14.04
  - ROS version: Jade
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

jraclarke, 2018-06-26:
It's possible that the points are being successfully loaded though at a different x, y, z to where you map is centred.  Are you also loading a tf launch file that corresponds to your coordinate reference frame?

Could you post the first 20 or so lines of one of the PCD files, so I can see if there are any other issues?

Thanks,

StanleyHusai, 2018-06-27:
@jraclarke yes i have loaded tf launch file, and open default.rviz in Rviz. But there is no map shown in rviz.
[image]
[image]

Here is some lines of PCD file
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 1135634
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 1135634
DATA ascii
-9.0422115 38.788029 -11.70006 12
-11.045238 38.801472 -11.763306 18
-10.920633 38.803806 -11.762435 18
-10.802614 38.801487 -11.760142 18
-10.996817 38.871117 -11.400371 17
-10.882156 38.874916 -11.399907 17
-10.767053 38.872719 -11.397462 17
-10.659155 38.875324 -11.396914 17
-10.030938 38.858513 -11.384604 11
-9.9163399 38.855892 -11.382892 22
-9.8088207 38.858204 -11.383076 16
-9.2569332 38.843204 -11.376274 16
-9.1496973 38.847446 -11.377802 22
-9.0422029 38.845734 -11.377344 16

closed 2018-07-02 by StanleyHusai

vimalrajayyappan, 2019-10-31:
Can you please say how did you load it?