autowarefoundation/autoware #1367: Gazebo does not launch

AthpapAutowareUser, 2018-07-10:
Hi.

When I try to open the gazebo from the Run time Manager launch button I get the below message in the terminal:
[catvehicle_skidpan.launch] is neither a launch file in package [catvehicle] nor is [catvehicle] a launch file name

The files do exist and of course I have run the source command : source /setup.bash before.
Even the relative file has as an internal comment : in order to launch it type:

roslaunch catvehicle catvehicle_skidpan.launch

Even directly with the command typed in the terminal I get the same error. Can anyone help here ?

yunjeongkim, 2018-07-12:
Hi, 
Go to /Autoware/ros/src/system/gazebo/catvehicle/ 
and delete CATKIN_IGNORE file and rebuild Autoware (./catkin_release). 
That might solve your problem.

AthpapAutowareUser, 2018-07-13:
What I did was to delete the CATKIN_IGNORE file and just restart the machine. The launches, but this time I have crash with the Rviz (not with a combination gazebo-rviz).

I have closed the gazebo and I would like to run a demo simulation now (with the usual sample files). In that case, when I launch the localization file , at that point, Rviz crashes.
Before everything was fine. Any idea ?

AthpapAutowareUser, 2018-07-15:
I found the the problem for the crash. 
Thank you very much for the help. just one question.
The CATKIN_IGNORE was an empty file. Why this caused a problem ?