autowarefoundation/autoware #1368: Autoware Source build is not working with Drive PX2

AkshaySDesai, 2018-07-11:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug

Despite meeting all the prerequisites for the source build specified in this website: 

https://github.com/CPFL/Autoware/wiki/Source-Build

At the command line: 

**rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO**

I encountered the following error: 

[image]

The error report is as follows:

[autoware_error.txt](https://github.com/CPFL/Autoware/files/2184808/autoware_error.txt)

## Expected Behavior

The command was just supposed to run the code and setup the catkin work-space. Any help in this regard is appreciated.

bdholt1, 2018-07-30:
@kfunaoka The error report attached does not contain any errors.  Are you sure it's the complete file?

ajay1606, 2018-12-05:
@AkshayDesaiSDCE & @bdholt1, I am also getting build errors, while trying to install autoware from the source. Could you please help me to fix.

Please check PR(#1745 ) (source build error without log information #1745)

Kind regards,
Ajay

closed 2020-05-29 by gbiggs