autowarefoundation/autoware #1371: Source Build error, please help

jt387, 2018-07-12:
Hi all,   I installed the ROS indigo,  and build Autoware refered to [https://github.com/CPFL/Autoware/wiki/Source-Build](url),  when executing the command "./catkin_make_release" gets error as below. I search on the WEB, but didn't find useful information. Please help me, thank a lot!

# [Bug]
## Expected Behavior
No build error

## Actual Behavior
-- +++ processing catkin package: 'lidar_localizer'
-- ==> add_subdirectory(computing/perception/localization/packages/lidar_localizer)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_SURFACE
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_IO
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_GEOMETRY
-- looking for PCL_KEYPOINTS
-- looking for PCL_REGISTRATION
-- looking for PCL_SEGMENTATION
-- looking for PCL_RECOGNITION
-- looking for PCL_VISUALIZATION
-- looking for PCL_PEOPLE
-- looking for PCL_OUTOFCORE
-- looking for PCL_TRACKING
-- looking for PCL_APPS
-- Eigen found (include: /usr/include/eigen3)
-- Using these message generators: gencpp;geneus;genlisp;genpy
CMake Error at /home/seacloud/Autoware/ros/devel/share/ndt_cpu/cmake/ndt_cpuConfig.cmake:141 (message):
  Project 'lidar_localizer' tried to find library 'DummyDevice'.  The library
  is neither a target nor built/installed properly.  Did you compile project
  'ndt_cpu'? Did you find_package() it before the subdirectory containing its
  code is included?
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:87 (find_package)
  computing/perception/localization/packages/lidar_localizer/CMakeLists.txt:36 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/seacloud/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/seacloud/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

## Specifications

  - Ubuntu version: 14.04
  - ROS version: indigo
  - Autoware branch:

jt387, 2018-07-13:
I tried the 1.7.0 version,  and also got the same error. Please do me a favor, thanks!

vingtfranc, 2018-07-17:
Same issue here.

vingtfranc, 2018-07-17:
This is an issue with OpenNI2. I have tried many different things. The path /usr/lib/OpenNI2/Drivers seems not to be taken into account for some reasons. A dirty fix that worked for me is to cp the .so files in /usr/lib/ (you should have a libDummyDevice.so among them).

I don't know if that's important but while I was struggling to fix the issue I also recompiled OpenNI2 from https://github.com/occipital/OpenNI2 and used the files from ./Bin/x64-Release/* and ./Include/ (the installation script seems broken). So you can also explore this solution if my first suggestion doesn't work.

kitsukawa, 2018-07-17:
This error occurs in the environment of Ubuntu 14.04, ROS Indigo, **pcl 1.7.2** . I met the error before and downgrading the pcl from 1.7.2 to 1.7.1 fixes this error. But some functions of Autoware 
(such as ndt_anh) are disabled when pcl 1.7.1 is installed. I have not found the fundamental solutions for this yet. On Ubuntu 16.04 + ROS Kinetic, this does not reported even if pcl 1.7.2 is installed.

jt387, 2018-07-23:
@vingtfranc  I copy the .so files into /usr/lib and solve the problem, thanks a lot!  I find that there is  the command "export LIBRARY_PATH=/usr/lib/OpenNI2/Drivers:$LIBRARY_PATH" in the catkin_make_release file, but it maybe doesn't work. 
@kitsukawa Thank you for reply! It's valuable!

closed 2018-07-23 by jt387

Srivatsa-M-Hegde, 2018-10-17:
@jt387 I seem to have a similar issue with cmake. I get the following error

Make Error at util/packages/sound_player/CMakeLists.txt:9 (find_package):
  By not providing "Findsound_play.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "sound_play", but CMake did not find one.

  Could not find a package configuration file provided by "sound_play" with
  any of the following names:

    sound_playConfig.cmake
    sound_play-config.cmake

  Add the installation prefix of "sound_play" to CMAKE_PREFIX_PATH or set
  "sound_play_DIR" to a directory containing one of the above files.  If
  "sound_play" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/srivatsa/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/srivatsa/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

so the only difference is the package configureation file for SOUND_PLAY  cannot be located. I am unable to locate the OpenNI2 folder in the /usr/lib folder for some reason. Is this normal ? also please elaborate on .so files that you copy to the /usr/lib folder since I see a bunch of .so files already in there.

yuqli, 2018-11-06:
I have the same error. Am looking into why. I use Ubuntu 16.04 + ROS kinetic.

yuqli, 2018-11-06:
> I have the same error. Am looking into why. I use Ubuntu 16.04 + ROS kinetic.

This particular error is resolved after installing sound-play using apt-get. Link here : https://answers.ros.org/question/260933/how-to-install-sound_play/ 

Command is `sudo apt-get ros-kinetic-sound-play`

But I'm facing a new error with jsk_recognition_msgs and will continue looking into why.