autowarefoundation/autoware #1374: Source Build Error,Please Help

PejoyPOcky, 2018-07-13:
I followed https://github.com/CPFL/Autoware/wiki/Source-Build to install Autoware. 
After installation, I can open Autoware, but can't find some packages like  lidar_localizer  . When  using ndt_mapping, there will be process died message  in the terminal.
And I found there are warnings when executing ./catkin_make_release,  it seems I don't have some libraries to compile the whole Autoware package.

# Bug
## Expected Behavior
Fully Build

## Actual Behavior
```
Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.
```

## Steps to Reproduce the Problem

  1.`./catkin_make_release`

## Specifications

  - Ubuntu version:16.04
  - ROS version:kinetic
  - Autoware branch:master

jt387, 2018-07-13:
It can't find the vtk binary and python wrapper libvtkRenderingPythonTkWidgets.so files,  you can try the commands: sudo apt-get install libvtk; sudo apt-get install python-vtk. Good luck!
And I followed the docker build:  [https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker](url),  and chose "Case 1: Using Docker Hub "As Is"", then build sucessfully.  Maybe you can have a try.

PejoyPOcky, 2018-07-15:
@jt387 Thanks for your advice, and I tried "Case1: Using Docker Hub As IS ".
But when executing `sudo sh run.sh kinetic `,  I always get  `nvidia-docker | 2018/07/15 01:32:56 Error: unexpected EOF` in terminal.
What is going on ??

bdholt1, 2018-07-31:
Hi @PejoyPOcky is this issue now resolved?

PejoyPOcky, 2018-08-05:
@bdholt1 as I commented before, I've installed autoware by following docker build case2.
but I  can't figure out why error occurs when using source build or docker build case 1.

bdholt1, 2019-01-23:
@PejoyPOcky during the past 6 months we've done a whole lot of cleanup on the docker files and this work is ongoing.  If your immediate problem is resolved then please close this issue.