autowarefoundation/autoware #1377: [bug] Static tf broadcaster in ndt_mapping

alexampa, 2018-07-17:
# Bug
## Expected Behavior
Create points map from velodyne lidar in Gazebo using ndt_mapping node

## Actual Behavior
PointCloud data are published from the lidar sensor and points_callback() function of ndt_mapping node is called. However br.sendTransform() command is not executed and thus transform from map frame to base_link frame is not published.

## Steps to Reproduce the Problem

  1. Launch Gazebo with a lidar sensor
  2. Launch ndt_mapping node

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Gazebo version: 7
  - Autoware branch: master

alexampa, 2018-07-17:
## Fix
The problem was solved to me by declaring tf::TransformBroadcaster object as static into points_callback() body. I made a pull request for that #1376

kfunaoka, 2018-08-08:
Merged

closed 2018-08-08 by kfunaoka