autowarefoundation/autoware #1383: grasshopper3_camera 0 camera detected

vingtfranc, 2018-07-20:
Hi,

I have installed Autoware from source and I am trying to run the demo (https://github.com/CPFL/Autoware/wiki/Demo)
Compilation goes fine, I can start the runtime manager, load the rosbag, start rviz, load the map (like in the demo video). However, when I launch the Sensing node, it crashes right away.  

# Bug
## Expected Behavior
Sensing node loaded

## Actual Behavior
Here's the whole console output I get:

```
loading param.yaml
loading qs.yaml
loading setup.yaml
['rosparam', 'set', 'tf_x', '1.2']
['rosparam', 'set', 'tf_y', '0.0']
['rosparam', 'set', 'tf_z', '2.0']
['rosparam', 'set', 'tf_yaw', '0.0']
['rosparam', 'set', 'tf_pitch', '0.0']
['rosparam', 'set', 'tf_roll', '0.0']
['rosparam', 'set', 'localizer', 'velodyne']
loading map.yaml
loading sensing.yaml
loading computing.yaml
loading interface.yaml
loading data.yaml
loading simulation.yaml
['rosparam', 'set', '/use_sim_time', 'false']
loading status.yaml
loading state.yaml
loading topics.yaml
['rosparam', 'set', '/use_sim_time', 'true']
['rosbag', 'play', '--clock', '/home/xxx/Autoware/rosbag/sample_moriyama_150324.bag']
pid=24410
sched policy=OTHER prio=0
['sh', '-c', '$(rospack find runtime_manager)/../../../.config/rviz/cmd.sh']
pid=24457
sched policy=OTHER prio=0
['roslaunch', '/home/xxx/Autoware/docs/quick_start/my_map.launch']
pid=24938
sched policy=OTHER prio=0
... logging to /home/xxx/.ros/log/cb4c00a4-8bbe-11e8-a177-ac220b522049/roslaunch-xxx-24938.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xxx:36362/

SUMMARY
========

PARAMETERS
* /rosdistro: indigo
* /rosversion: 1.11.21

NODES
/
map_to_mobility (tf/static_transform_publisher)
points_map_loader (map_file/points_map_loader)
vector_map_loader (map_file/vector_map_loader)
world_to_map (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
load /home/xxx/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00864.pcd
load /home/xxx/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00852.pcd
load /home/xxx/.autoware/data/map/pointcloud_map/bin_Laser-00159_-00859.pcd
[... 106 lines cut ...]

core service [/rosout] found
[ INFO] [1532051300.271912198]: fps set to 20
[ INFO] [1532051300.272625906]: mode set to 0
[ INFO] [1532051300.273300782]: format set to raw
[ INFO] [1532051300.274017982]: timeout set to 1000 ms
Number of cameras detected: 0
Error: This program requires at least 1 camera.
[grasshopper-4] process has died [pid 25305, exit code 255, cmd /home/xxx/Autoware/ros/devel/lib/autoware_pointgrey_drivers/grasshopper3_camera __name:=grasshopper __log:=/home/xxx/.ros/log/cb4c00a4-8bbe-11e8-a177-ac220b522049/grasshopper-4.log].
log file: /home/xxx/.ros/log/cb4c00a4-8bbe-11e8-a177-ac220b522049/grasshopper-4*.log

```

Note that I don't have any file named grasshopper-4.log in ~/.ros/log/*

## Steps to Reproduce the Problem

I followed the installation and demo guide.
I downloaded and installed flycapture2-2.9.3.43-amd64 in order to compile grasshopper3_camera

## Specifications

  - Ubuntu version: 14.04
  - ROS version: Indigo
  - Autoware branch: 1.7.0

amc-nu, 2018-07-20:
@vingtfranc seems you have an incorrect camera setup.

> Number of cameras detected: 0

Please confirm that the camera is correctly detected using the FlyCap tool included in FLIR's SDK. If it is not detected, follow FLIR's setup guide for the camera.

closed 2018-07-20 by amc-nu

vingtfranc, 2018-07-20:
I'm not sure I understand. Do you mean I need to have a real camera in order to feed Autoware with the demo rosbag?

amc-nu, 2018-07-20:
If you’re using a rosbag then there is no need to launch the camera node. The image data is already included in the demo rosbag.

kfunaoka, 2018-07-20:
Thanks Abraham!

@vingtfranc You can reproduce the demo even if you ignore the crash of grasshopper. The camera node is not used in the simulation. It may be confusing. It seems that my_sensing.launch is dummy.

vingtfranc, 2018-07-20:
It seems the demo rosbag does not contain camera data (/image_raw right?):
```
$ rosbag info ~/Autoware/rosbag/sample_moriyama_150324.bag 
path:        /home/xxx/Autoware/rosbag/sample_moriyama_150324.bag
version:     2.0
duration:    7:59s (479s)
start:       Mar 24 2015 01:40:50.33 (1427157650.33)
end:         Mar 24 2015 01:48:49.50 (1427158129.50)
size:        7.9 GB
messages:    16768
compression: none [4788/4788 chunks]
types:       nmea_msgs/Sentence      [9f221efc5f4b3bac7ce4af102b32308b]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /nmea_sentence   11980 msgs    : nmea_msgs/Sentence     
             /points_raw       4788 msgs    : sensor_msgs/PointCloud2
```

kfunaoka, 2018-07-20:
Right.
/image_raw is removed because it is heavy and not used in the demo.

vingtfranc, 2018-07-20:
Thank you, I'll download the original from the rosbag store then

vingtfranc, 2018-07-20:
I realized that I cannot download /image_raw data on the rosbag store without a premium account. Considering I'm just wanting to make a demo, is there any rosbag (even small) available for free?