autowarefoundation/autoware #1384: [Feature] Refactor NDT functions as NDT SLAM

kfunaoka, 2018-07-20:
# New Feature
NDT functions have multiple implementations.
They can be merged into a single implementation as NDT SLAM node.
NDT SRAM node can execute NDT mapping and/or matching simultaneously or individually.

NDT SRAM node will be only the implementation of NDT if it is proven to work correctly enough.
The existing NDT nodes such as ndt_matching and ndt_mapping can be used for a while of course.

alexampa, 2018-08-02:
@kfunaoka It would be also desired if you added a parameter for the base_link frame so that the user can add the frame name he wants for the transformation, instead of being hard coded, as implemented in the [gmapping](https://github.com/ros-perception/slam_gmapping/blob/5a707e079364ab9f5349c72310f31bfed4a6c834/gmapping/src/slam_gmapping.cpp#L789) slam node. I'm referring to the command
```
br.sendTransform(tf::StampedTransform(transform, current_scan_time, "map", "base_link"));
```
in points_callback() function inside ndt_mapping and ndt_matching nodes. Thanks in advance! :)

kfunaoka, 2018-08-08:
@alexampa Thank you for the feedback! Yamato will take care of that.

alexandrx, 2019-04-01:
@kfunaoka I understand the plan is to merge `ndt_mapping` and `ndt_matching` into one node `NDT SLAM`. But why do you need "SLAM" word in the refactoring? What do you plan to do with `ndt_{mapping,matching}_TKU` and `approximate_ndt_mapping`?

Also, as you may know, there is the ICP-based mapping 
http://wiki.ros.org/ethzasl_icp_mapper
therefore if we follow your proposal, and if we add the above ETHZ-ASL ICP mapper to Autoware, then we need to refactor both icp matching and mapping into some `ICP SLAM`

Maybe be keep it simple.

kfunaoka, 2019-04-17:
@alexandrx `NDT SLAM` can also do mapping and matching simultaneously. As far as I know, there is no plan regarding `ndt_{mapping,matching}_TKU`, `approximate_ndt_mapping`, and `ICP SLAM` for now. If there are any reasons that prioritize them, we should do that :)

gbiggs, 2019-04-17:
> therefore if we follow your proposal, and if we add the above ETHZ-ASL ICP mapper to Autoware, then we need to refactor both icp matching and mapping into some ICP SLAM

What refactoring would be needed? Why can't we use the `ethzasl_icp_mapper` package directly?

closed 2020-05-29 by gbiggs