autowarefoundation/autoware #1388: Build error  "module_comm_msgs"

andoh104, 2018-07-22:
# Bug
## Expected Behavior

No Error

## Actual Behavior

```
-- +++ processing catkin metapackage: 'velodyne'
-- ==> add_subdirectory(sensing/drivers/lidar/packages/velodyne/velodyne)
-- +++ processing catkin package: 'velodyne_msgs'
-- ==> add_subdirectory(sensing/drivers/lidar/packages/velodyne/velodyne_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- velodyne_msgs: 2 messages, 0 services
-- +++ processing catkin package: 'as'
-- ==> add_subdirectory(actuation/vehicles/packages/as)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'module_comm_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "module_comm_msgs"
  with any of the following names:

    module_comm_msgsConfig.cmake
    module_comm_msgs-config.cmake

  Add the installation prefix of "module_comm_msgs" to CMAKE_PREFIX_PATH or
  set "module_comm_msgs_DIR" to a directory containing one of the above
  files.  If "module_comm_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  actuation/vehicles/packages/as/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/tandoh/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/tandoh/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
:~/Autoware/ros$ 
```

## Steps to Reproduce the Problem

  1.`./catkin_make_release` 
  1.
  1.

## Specifications

  - Ubuntu version:16.04
  - ROS version:kinetic
  - Autoware branch:master

This file cause this error:
https://github.com/CPFL/Autoware/blob/0a48443ebcf3fa9ec735191d8d549d61762ede8c/ros/src/actuation/vehicles/packages/as/CMakeLists.txt

After this commit this error occur:
https://github.com/CPFL/Autoware/commit/0a48443ebcf3fa9ec735191d8d549d61762ede8c#diff-d9b74b46302f7d4b38a927daac2753a2

andoh104, 2018-07-22:
@amc-nu 
Could you check the error and this commit ?

https://github.com/CPFL/Autoware/commit/0a48443ebcf3fa9ec735191d8d549d61762ede8c#diff-d9b74b46302f7d4b38a927daac2753a2

amc-nu, 2018-07-22:
Hello @andoh104 

I think you’re missing some steps to build
https://github.com/CPFL/Autoware/wiki/Source-Build

Starting from
`git submodule update --init --recursive`

andoh104, 2018-07-22:
@amc-nu
Thank you very much.

I was able to fix this by your way.
And I found this is same as https://github.com/CPFL/Autoware/issues/1301 . But it was only in docker.

closed 2018-07-22 by andoh104