autowarefoundation/autoware #1393: run demo error,  please help

jt387, 2018-07-23:
Hello everyone!  I have installed the Autoware from the source build, and I want to run the demo: [https://github.com/CPFL/Autoware/wiki/Demo](url). I execute the "./run" command,  there are two windows: the roscore window and the input password window. 
 What is the password?  After I input the password of the user logged in Ubuntu,  the input  password window exits and Autoware Runtime Manager window doesn't  appear.  

# Bug
## Expected Behavior
Autoware Runtime Manager window 

## Actual Behavior
No Autoware Runtime Manager window 

## Specifications

  - Ubuntu version:14.04
  - ROS version: indigo
  - Autoware branch: master

kfunaoka, 2018-07-23:
@jt387 Does the demo succeed when using Docker Case 1: Using Docker Hub "As Is"?
https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker

jt387, 2018-07-23:
@kfunaoka Unfortunately, I run the demo also failed using the Docker Case 1, and I have openned another issue.