autowarefoundation/autoware #1394: The docker build version:  Autoware Runtime Manager  can't  display  or find the  .autoware  directory

jt387, 2018-07-23:
Hello everyone！  I have installed the Autoware from the docker build, and I want to run  the demo: [https://github.com/CPFL/Autoware/wiki/Demo](url). Firstly, I executed the command as below:

seacloud@dell-Precision-Tower-5810:~/Autoware/docker/generic$ sudo sh run.sh indigo /home/seacloud/Autoware/shared_dir
[sudo] password for seacloud: 
Use indigo
Shared directory: /home/seacloud/Autoware/shared_dir
autoware@dell-Precision-Tower-5810:~$ pwd
/home/autoware

And the user turns to autoware from seacloud,  after I execute the "./run" under the /home/autoware/Autoware/ros directory,  the roscore window, the Terminal window which displays loading the .yaml and the Autoware Runtime Manager window appear, but there is no input password windows.  I create the .autoware directory under the /home/root,  and I download the Demo data into the .autoware directory.  But  the Autoware Runtime Manager's Simulation window can't find the .autoware directory,  and I can't copy the .autoware into the /home/autoware.  

My question is which directory should I put the .autoware,  and why the Autoware Runtime Manager 's  Simulation window can't find the .autoware?  Please help me! Thanks! 
 

# Bug
## Expected Behavior
Simulation can find the .autoware directory and load the sample ROSBAG data.

## Actual Behavior
Simulation doesn't find the .autoware directory.

## Specifications

  - Ubuntu version:14.04
  - ROS version:indigo
  - Autoware branch:master

jt387, 2018-07-23:
I don't know why there is the underlines between the fourth line and the ninth line.

jt387, 2018-07-23:
I don't know why there is the underlines between the fourth line and the ninth line on the top

kfunaoka, 2018-07-23:
@jt387 You can use /home/$USER/shared_dir for data sharing.

kfunaoka, 2018-07-23:
Check how to use general Docker and see docker/generic/run.sh.
```
nvidia-docker run \
    -it --rm \
    --volume=$XSOCK:$XSOCK:rw \
    --volume=$XAUTH:$XAUTH:rw \
    --volume=$HOST_DIR:$SHARED_DIR:rw \                <---- here
    --env="XAUTHORITY=${XAUTH}" \
    --env="DISPLAY=${DISPLAY}" \
    -u autoware \
    --privileged -v /dev/bus/usb:/dev/bus/usb \
    --net=host \
    autoware-$1
```

YangLiu14, 2019-04-25:
Hi i have the same Gtk-Warning when loading the map by following the Demo, did you solve this problem now?

closed 2020-05-29 by gbiggs