autowarefoundation/autoware #1401: Question about imm_ukf_pda node

Tpsych, 2018-07-27:
Hey guys:

In imm_ukf_pda.cpp, what does the variable **det_explode_param** stand for?
I thought determinant is used to calculate area/volume, what's the actual meaning here ??
Not good at math, it would be great if any math expert can answer ！

---------------------------------------------------------------------------------------------------------------
void ImmUkfPda::tracker(const autoware_msgs::CloudClusterArray& input,
                        jsk_recognition_msgs::BoundingBoxArray& jskbboxes_output,
                        autoware_msgs::DetectedObjectArray& detected_objects_output)
{
  double timestamp = input.header.stamp.toSec();

  double **det_explode_param** = 10;
  double cov_explode_param = 1000;

cirpue49, 2018-07-30:
I am not math expert, but det_explode_param stands for a rule-based threshold for preventing UKF from exploding.

If you do not set this param, you would get unexpected behavior/error like segmentation fault.

Yes, determinant represents area/volume. In this algorithm, UKF stops tracking if covariance's volume exceeds the certain threshold.

Tpsych, 2018-08-01:
@cirpue49 Thanks！

kfunaoka, 2018-08-01:
Answered.

closed 2018-08-01 by kfunaoka