autowarefoundation/autoware #1406: Release 1.8.0

kfunaoka, 2018-07-30:
I’ll create release/1.8.0 branch on the morning of Aug. 6th.
After that the commits that should be included in v1.8.0 will be cherry-picked if possible.

esteve, 2018-08-08:
@kfunaoka do you think https://github.com/CPFL/Autoware/pull/1398 could be added to the milestone now that it's merged? I've submitted https://github.com/CPFL/Autoware/pull/1429 to make it easier to cherry pick it.

kfunaoka, 2018-08-09:
@esteve Yes. I think it's better to add #1398 to v1.8. 

I'll do the following action items.
- autoware/crossbuild:aarch64-kinetic-1.0.0 on DockerHub can be removed.
- autoware/crossbuild is remamed into autoware/build for using Docker with self build mentioned at https://github.com/CPFL/Autoware/pull/1396#issuecomment-409539642.
- autoware/build:generic-aarch64-YYYYMMDD and autoware/build:synquacer-YYYYMMDD will be pushed.
- Travis will use autoware/build:generic-aarch64-YYYYMMDD.

esteve, 2018-08-17:
I've submitted https://github.com/CPFL/Autoware/pull/1452 in case it could be added to the 1.8 release more easily.

kfunaoka, 2018-08-29:
Updated with URIs of Video and Readme

closed 2018-08-31 by kfunaoka