autowarefoundation/autoware #1409: [Fix] Rename types and topics in Detection

kfunaoka, 2018-07-31:
# New Feature
The names of types and topics in Autoware do not seem uniformed. We should discuss the naming rules of types and topics. We'd like to fucus on Detection in [Overview](https://github.com/CPFL/Autoware/wiki/Overview) at first.

kfunaoka, 2018-07-31:
I've created a 1st draft of component diagram.
The level of abstraction is same as [Overview](https://github.com/CPFL/Autoware/wiki/Overview).
Please focus on Detection, left side in the figure.

@shinpei0208 @amc-nu Would you comment about it?

[image]

Generated from https://github.com/CPFL/Autoware/tree/feature/architecture/docs/architecture

LukePhairatt, 2018-07-31:
Hi Kinji, I got some ideas and hope it is somewhat useful.
1- Including ‘Array’ at the end of the message type as indicated in the diagram.
    ‘ObjectArray’

2- Including ‘Array’ at the end of the topic name. ‘/detected_objects_vision_array’ but limited to 4    
    words
    
    These 1&2 will align with what already used in Autoware for example
    Message Type: autoware_msgs/LaneArray
    Topic Name: "/lane_waypoints_array"

3- ‘autoware_msgs’ namespace
    Could we be more specific about the message type as defined in ROS? 
    For example ‘geometry_msgs/TwistStamped’
    so we can define autoware specific messages to something like ‘autoware_(xxx)_msgs/xxxx’
    for example ‘autoware_paths_msgs/LaneArray’
    or something along this idea.            

4- Capital word of the type name such as geometry_msgs/TwistStamped 
    Not all are like that at the moment such as ‘autoware_msgs/lane’

5- ‘Topic’ naming convention?
     for example /ParentNode_ChildNode_DataName
                         /detection_vision_object_array
                         /detection_lidar_object_array
                         /detection_fusion_object_array
                         /detection_combined_object_array
                         /detection_tracked_object_array
 
     or else relative namespace e.g. detection/vision_object_array

kfunaoka, 2018-08-01:
@LukePhairatt Thank you for the beneficial advice. I've updated the component diagram as 2nd draft thanks to you.

[image]

> 2- Including ‘Array’ at the end of the topic name. ‘/detected_objects_vision_array’ but limited to 4
words

I think “_array” at the end of topic name is not a good idea because it simply represents type. Similar to dataname_int. So I removed the “_array” in this 2nd draft.

kfunaoka, 2018-08-01:
Because object_tracker requires one ObjectArray as input, object_tracker inputs only detected_objects. detected_objects can be selected from one of vision_objects, lidar_objects, combined_objects, or fused_objects.

LukePhairatt, 2018-08-01:
@kfunaoka Looks good. Yes agreed with you on this. "_array” at the end of topic name is not a good idea because it simply represents type."

I proposed this _array concatenation just because it will in line with other topics such as "lane_waypoints_array". But it is better without it.

kfunaoka, 2018-08-01:
@LukePhairatt Thanks! I'd like to ignore the current implementation at the spec level if the modification is not too much.

esteve, 2018-08-01:
@kfunaoka @LukePhairatt I've been working on splitting the `autoware_msgs` package to better understand the inter-package dependencies and how the ROS nodes talk to each other. You can have a look at https://github.com/esteve/Autoware/tree/fix/split_autoware_msgs

kfunaoka, 2018-08-01:
@esteve That's great! It brings the visibility of Autoware internal.

@yk-fujii @aohsato @amc-nu @kitsukawa @YamatoAndo @cirpue49 @yn-mrse 
If you have any comment, please let me know. I think it will not be a problem because his modification is only where autoware_msgs exists.

kfunaoka, 2018-08-01:
> I think it will not be a problem because his modification is only where autoware_msgs exists.

In the viewpoint of conflicts with your work.

kfunaoka, 2018-08-02:
I've write down the current conclusion at the headline https://github.com/CPFL/Autoware/issues/1409#issue-346099272.

kfunaoka, 2018-08-08:
I've created the page of design rules.
https://github.com/CPFL/Autoware/wiki/Design-Rules

kfunaoka, 2018-10-15:
I'd like to update naming convention of topic names as below.

1. We will use hierarchical namespace inside each function in [Overview](https://github.com/CPFL/Autoware/wiki/Overview). For example, the naming convention inside Detection is `/detection/ParentComponent/DataName`. For example, `/detection/image_detector/objects` is used between `image_detector` and `image_tracker`. 
1. I/O among functions in [Overview](https://github.com/CPFL/Autoware/wiki/Overview), e.g., Detection and another function such as Prediction, `/detection/objects` is used.

The previous naming convention `/namespace/ParentComponent_DataName` might be difficult to read if there are a lot of data names and/or some of them are long.

Let me know if any feedback.

[image]

closed 2020-05-29 by gbiggs