autowarefoundation/autoware #1410: is NDT_mapping real time?

scalefreee, 2018-08-01:
When I played the .bag, the ndt mapping seems to be lagging behind, observed from RViz. When the .bag finished playing, the map is only partially done and stopped. 

I am wondering is it because the computing speed does not match the .bag replay speed? or what is the mechanism in the mapping process.

Thanks.

kitsukawa, 2018-08-01:
@scalefreee Regarding the first question, mapping process is not real time, so mapping is behind the rosbag play. Regarding the second question, I don't think it is due to low resolution of lidar. We have created maps with certain accuracy using ndt_mapping from Velodyne VLP-16 data. I think it is due to few 3d feature required to perform 3D scan matching or not appropriate parameter tuning.

scalefreee, 2018-08-02:
thanks a lot for the answer.

We tried another bag recording the same area using the same lidar, and the mapping is working now. Haven't figured out what is the difference between two bags.

btw, is there any detailed description for the parameters used in ndt_mapping and ndt_matching modules? Thanks @kitsukawa