autowarefoundation/autoware #1428: [Fix] Split autoware_msgs

kfunaoka, 2018-08-08:
# New Feature
All Autoware-dedicated messages belong to autoware_msgs. This issue splits autoware_msgs based on [Design Rules](https://github.com/CPFL/Autoware/wiki/Design-Rules) to better understand the inter-package dependencies and how the ROS nodes talk to each other.

kfunaoka, 2018-08-08:
@esteve I've created the issue you mentioned at https://github.com/CPFL/Autoware/issues/1409#issuecomment-409625199. Is it OK to assign it to you?
https://github.com/esteve/Autoware/tree/fix/split_autoware_msgs

esteve, 2018-08-08:
@kfunaoka yeah, it's ok, thanks!

alexandrx, 2019-02-23:
We've been working on `autoware_msgs` refactoring. I'm almost done with new organization of namespaces and currently running tests (still some manual adjustments). 

Here is the document describing the work so far (`feature/autoware_msgs_refactoring` branch):
https://github.com/CPFL/Autoware/blob/feature/autoware_msgs_refactoring/ros/concerning-autoware-msgs.md

Kindly check whether it satisfies what you expected.

alexandrx, 2019-03-01:
Regarding `autoware_msgs` refactoring, I got some feedback. There are some minor changes as to the plan and message packages.

First, the changes:
1. Merge `navigation` and `decision` messages, and rename that as `planning`.
2. `IndicatorCmd` and `LampCmd` are exactly the same thing (contents), replace `IndicatorCmd` for `LampCmd`.
3. Move `VehicleCmd` into the new `planning` namespace (see 1)
4. Some objections regarding `socket` namespace (i.e., `autoware_socket_msgs`). It currently holds the `RemoteCmd` message advertised by "mqtt_receiver.cpp" (which lives in `src/socket/packages/mqtt_socket`). If there are no other objections I still will keep `autoware_socket_msgs`.
```
```
Second, the plan:
For each category `C` in [`detection`, `localization`, `planning`, `sensing`, `sync`, `socket`]:
1. Create the new message package `autoware_C_msgs` with its corresponding message files (move the message files out of `autoware_msgs` into `autoware_C_msgs`)
2. Keep the old `autoware_msgs` package (except those files that went to `autoware_C_msgs`)
3. Create a new branch based on develop called `fix/refactoring_C_msgs`
4. Apply the refactoring script to change `autoware_msgs` into `autoware_C_msgs` only to those files which are using messages in `autoware_C_msgs`
5. Build Autoware on branch `fix/refactoring_C_msgs`, then issue the PR for that branch
6. You would be able to merge and fully build each PR.
```
```
Some other concerns/comments:
1. Some messages are used without referring the `autoware_msgs` message type. Ex: in "twist_gate.cpp" -> `state_cmd_pub_ = nh_.advertise<std_msgs::Int32>("/state_cmd", 1, true);` *BUT* there is already `autoware_msgs::StateCmd` (which has header and a std_msgs::Int32). While the refactoring won't affect such cases, it's preferable to use the existing message (or delete the message type if nothing is using it)
2. My script searches for the publishers of each message type as a hint for message namespaces name and message type association. However, it wasn't designed to find type aliases, ex. in "twist_gate.cpp" -> `using remote_msgs_t = autoware_msgs::RemoteCmd;` and also in "twist_gate.cpp" -> `using vehicle_cmd_msg_t = autoware_msgs::VehicleCmd;`. `VehicleCmd` was associated to an incorrect namespace `socket` while it should have been `planning`. Fortunately this aliasing is limited to "twist_gate.cpp". I don't suggest to prohibit type aliases, but using directly the autoware message without the alias makes the code easier to read.

gbiggs, 2019-03-01:
Here are my comments on the document prior to the most recent update, for completeness of this discussion.

- If a message is completely unused, we should delete it now, not make a package for it. If we need it again in the future we can grab it from the repository history.
- Please update your tables to match your new diagram.
- Is IndicatorCmd a utility message, or should we have a package for car-status-related messages? We have VehicleStatus as well, which is proposed to be in localization_msgs.
- Once the splitting up is done we need to consider renaming some of these messages to better match their purpose.
- ExtractedPosition contains information about a single bulb in a traffic light. Why is it in navigation_msgs? (It should probably be renamed, too.)
- What is LampCmd for and how is it different from IndicatorCmd?
- I'm not fond of the name "socket_msgs". Why did you choose this?
- The distribution of command-related messages between decision_msgs and navigation_msgs is not clear. Why are they done this way?
- In general justification for the location of each message would be good; it would make a PR easier to review and facilitate discussion.

gbiggs, 2019-03-01:
Here are some comments on the most recent edition.

- Many of the messages need to be renamed, such as `ExtractedPosition`. We should do this in a separate PR after this one is done, but I just want everyone to be aware that renaming will happen.
- Same goes for removing duplicate messages with different names, e.g. `LampCmd` and `IndicatorCmd`. We'll do these in a separate PR.
- We'll also need to put comments in the message definitions as well because some of them just have data fields with no clear purpose for the message coming from the name.
- Can and should we split messages related to data for planning, such as lanes and waypoints, from commands for controlling hardware?
- I would prefer to choose and name message locations based on the purpose of the message, not who publishes it. Treat messages as data types, because that is what they are. So `autoware_socket_msgs` doesn't tell me the purpose of the messages contained within, whereas `autoware_remote_control_msgs` does. (I'm not saying that's the right name for that package; I cannot tell what that message is used for based on its content or name.)
- The messages in `sensing_msgs` seem more about sensors than sensed data. Should the package be `autoware_sensor_msgs`?
- `VehicleStatus` does not contain localisation information. It talks about the state of the vehicle. It would do well in a package with other hardware-related messages. Perhaps the hardware control command messages.

The plan sounds good to me. Make sure, if possible, that each PR can be merged independent of the others so we do not have to merge them in the correct order.

alexandrx, 2019-03-01:
@gbiggs Thanks for the comments.
* As for the `autoware_sensing_msgs`, @amc-nu also mentioned this. However I respectfully disagree. Different from ROS `sensor_msgs` which are actually messages exchanged by sensors, `autoware_sensing_msgs` are for sensing (sensor data processing) tasks. So far such messages are limited to `CameraExtrinsic.msg`, `PointsImage.msg`, `ProjectionMatrix.msg` purely sensor data post-processing. While in ROS `sensor_msgs` you find the joystick, laser, pointcloud2, temperature, battery, etc., purely sensor data. We don't need to imitate ROS naming convention. But that's my opinion.
* As for your very last comment, I'm afraid we cannot do independent PRs in this case. There are several message inter-dependencies, each new branch will add new messages progressively replacing the old ones, and in the end we will remove old `autoware_msgs`. Cannot start testing with the PR which removes `autoware_msgs`.

gbiggs, 2019-03-03:
> As for the autoware_sensing_msgs, @amc-nu also mentioned this. However I respectfully disagree. Different from ROS sensor_msgs which are actually messages exchanged by sensors, autoware_sensing_msgs [...]

I wasn't mimicking the ROS core package name. I was thinking about the content of the messages in that package. If you want to avoid causing confusion with a major ROS package name, then I suggest `autoware_sensor_info_msgs`.

> As for your very last comment, I'm afraid we cannot do independent PRs in this case. There are several message inter-dependencies, each new branch will add new messages progressively replacing the [...]

You can, but it involves touching the content of the files as well as moving them. If we had lots of tests so we could be sure each change wasn't breaking things, I'd insist you do it that way, but since we don't I'm not going to push such an unreasonable demand. I just want you to consider doing so if you find it possible for one of the PRs.

closed 2020-05-29 by gbiggs