autowarefoundation/autoware #1445: ndt_mapping eigen throw  align erro,  use gpu.

zslzsl999, 2018-08-14:
Fill-out only one section depending on whether you are reporting a bug or a new feature.
##BActual Behavior

I wanted to generate pcd file by using ndt_mapping, but autoware stopped after 48 hours.
So I have to try ndt_mapping in GPU.
catkin_make works fine. The error throwout when I click ndt_mapping in "Computing".
It doesn't matter "pcl_generic" or "pcl_anh_gpu" were choosed in "Method Type".
I have check all the posibilities which described on the website "**** READ THIS WEB PAGE !!! ****" .

> SUMMARY
========

PARAMETERS
* /ndt_mapping/imu_topic: /imu_raw
* /ndt_mapping/imu_upside_down: False
* /ndt_mapping/incremental_voxel_update: False
* /ndt_mapping/method_type: 2
* /ndt_mapping/use_imu: False
* /ndt_mapping/use_odom: False
* /rosdistro: kinetic
* /rosversion: 1.12.13

NODES
/
ndt_mapping (lidar_localizer/ndt_mapping)
queue_counter (lidar_localizer/queue_counter)

ROS_MASTER_URI=http://localhost:11311

ndt_mapping: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[ndt_mapping-2] process has died [pid 14227, exit code -6, cmd /home/zsl/git/Autoware/ros/devel/lib/lidar_localizer/ndt_mapping __name:=ndt_mapping __log:=/home/zsl/.ros/log/c9fd825a-9f9f-11e8-93f1-9cb6d0faafad/ndt_mapping-2.log].
log file: /home/zsl/.ros/log/c9fd825a-9f9f-11e8-93f1-9cb6d0faafad/ndt_mapping-2*.log

## Specifications

  - Ubuntu version: Ubuntu 16.04.4 LTS
  - ROS version: kinetic
  - Autoware branch: 
`zsl@zsl-Alienware-15-R3:~/git/Autoware$ git branch
* master`
  - cuda version: 9.2
`zsl@zsl-Alienware-15-R3:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.10' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) `
`zsl@zsl-Alienware-15-R3:~$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.10' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 
`
`zsl@zsl-Alienware-15-R3:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17`
`zsl@zsl-Alienware-15-R3:~$ dpkg-query -s libeigen3-dev
Package: libeigen3-dev
Status: install ok installed
Priority: extra
Section: libdevel
Installed-Size: 5130
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Source: eigen3
Version: 3.3~beta1-2
Depends: pkg-config
Suggests: libeigen3-doc, libmrpt-dev
Description: lightweight C++ template library for linear algebra
 Eigen 3 is a lightweight C++ template library for vector and matrix math,
 a.k.a. linear algebra.
 .
 Unlike most other linear algebra libraries, Eigen 3 focuses on the simple
 mathematical needs of applications: games and other OpenGL apps, spreadsheets
 and other office apps, etc. Eigen 3 is dedicated to providing optimal speed
 with GCC. A lot of improvements since 2-nd version of Eigen.
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Homepage: http://eigen.tuxfamily.org
`

jyakaranda, 2018-10-10:
same error while using ndt_matching

daifuku07, 2018-10-10:
I had the same problem when recompiling using "catkin_make" command.
"catkin_make -DCMAKE_BUILD_TYPE=Release" is suitable to solve this problem.

jyakaranda, 2018-10-10:
> I had the same problem when recompiling using "catkin_make" command.
> "catkin_make -DCMAKE_BUILD_TYPE=Release" is suitable to solve this problem.

Thank you very much! Your suggestion solved my "unaligned_array" error.
But I encountered another error while using ndt_gpu on my [TX2](http://www.nvidia.com/object/embedded-systems-dev-kits-modules.html)(an arm development board) :
`Error: too many resources requested for launch /home/nvidia/workspace/source/Autoware/ros/src/computing/perception/localization/lib/ndt_gpu/src/VoxelGrid.cu 657`
This error is similar to #1006 , but i have not get a workaround for this error.

closed 2020-05-29 by gbiggs