autowarefoundation/autoware #1448: Multi lidar localization question

greenfield932, 2018-08-16:
Hi, 
is it possible to use ndt_mapping and ndt_matching with multiple LiDAR configuration?
For example, for large scale vehicle I can't just put LiDAR on top of it because of beams intersection with vehicle, but I can put LiDARs on each side of the vehicle. Is this case supported? If not, my intuition is to union all point clouds into one point cloud and probably make such transformations to each point cloud that they seems to belong to one virtual LiDAR in the center of the vehicle. Does it make sense? I highly appreciate any advice.

aohsato, 2018-08-17:
@greenfield932 Yes of course, please use `points_concat_filter` to merge and apply transform points provided from each lidar. You need pointcloud and `/tf` between each `frame_id`.
https://github.com/CPFL/Autoware/blob/master/ros/src/sensing/filters/packages/points_preprocessor/nodes/points_concat_filter/points_concat_filter.cpp

For now, we can use up to 8 lidars in develop branch. #1254

greenfield932, 2018-08-17:
@aohsato, thanks a lot!

closed 2018-08-17 by greenfield932