autowarefoundation/autoware #1453: For frame [velodyne] does not exist

hamkuchan, 2018-08-18:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
in RViz reports:
1:[velodyne] does not exist
2:Parameter [robot_description] does not exist and was not found by searchParameter.
## Expected Behavior
in Rviz should be a car

## Actual Behavior
there is no car

## Steps to Reproduce the Problem

  1. I followed the Autoware install guide and then the quick start, but I got the RViz without a car on it just like the guide video on youtube. And the map does not move at all.
  1.
  1.

## Specifications

  - Ubuntu version: 16.0.4
  - ROS version: kinetic
  - Autoware branch:Master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

closed 2018-09-07 by amc-nu

baroselfdriving, 2020-02-04:
i have the same problem!

Jaskaran-Singh-Sodhi, 2020-07-28:
Was this ever solved?

JWhitleyWork, 2020-07-28:
@Jaskaran-Singh-Sodhi This is a very old issue. If you are having a similar one, please post a question on ROS Answers (where we handle support requests).